dartsim/dart #618: Using osgDart in an external project

mkoval, 2016-02-22:
I am having some difficulty using `osgDart` in another project that includes DART via CMake. I looked at the examples in the `osgDart/examples` directory, but they are of limited usefulness because they are built via an `add_subdirectory` command in the main DART `CMakeLists.txt` file.

I can't figure out what I need to pass to `find_package(DART REQUIRED COMPONENTS ...)` to get the `osgDart` include directories and libraries. It's not as simple as adding `osgDart` to my `target_link_libraries` because I get this linking error:
```
undefined reference to symbol '_ZN9osgViewer4View17setUpViewInWindowEiiiij'
//usr/lib/libosgViewer.so.99: error adding symbols: DSO missing from command line
```

**Is there any documentation for how to use `osgDart` outside of DART itself?**

jeffeb3, 2016-02-22:
I had to add `osgDart` to `target_link_libraries` and I had to add `find_package(OpenSceneGraph 2.0.0 REQUIRED COMPONENTS osgDB osgUtil osgViewer)` and then add `${OPENSCENEGRAPH_LIBRARIES}` to `target_link_libraries`.

I'm not sure that's the right way, but it is working for me.

mxgrey, 2016-03-07:
Whenever I use DART in an outside application, I find myself using ``find_package`` on Eigen and OpenSceneGraph libraries. Is it kosher to set up ``find_package(DART ... )`` to find those packages itself and add them to the ``${DART_LIBRARIES}`` variable? If so, then perhaps we should do that.

jeffeb3, 2016-03-07:
You can run dart without open scene graph. You can't use dart without Eigen.

Really, there should be two packages, or components. 
1) DART should add the -I for eigen, since you need that to include the dart files.
2) osgDART should add the minimum amount of OSG to build against DART, but if you are using osgAnimation, that's on you to find_package.

AFAIK, there isn't the second target/component, (hence the direct link of osgDart above) but maybe I'm just not using it right.

mxgrey, 2016-03-07:
Right, OSG libraries would only be included if ``osgDart`` is requested as a component by the user. In the future ``osgDart`` will actually be a COMPONENT called ``dart::gui::osg``.

jeffeb3, 2016-03-07:
Then, yeah. I think it is "the right way" to have the dependencies bubbling up, but only the minimum needed for osgDart. If you have a .cpp file with only includes from osgDart, and nothing else, it should compile without needing to do any other find_packages.

mkoval, 2016-03-08:
I agree 100% with @jeffeb3 here. One `find_package` command should work with no additional shenanigans by the user.

I `grep`ed through the DART headers and found the following transitive dependencies that need to be added to `DART_INCLUDE_DIRS` and `DART_LIBRARIES` in the `DARTConfig.cmake` file:

- Assimp
- Boost (algorithm, math, lexical_cast, regex)
- Bullet
- coin
- Eigen
- FCL 
- OpenGL
- GLUT
- flann
- openscenegraph
- TinyXML
- TinyXML2
- NLopt
- urdf_model
- urdf_parser
- urdf_world

In some cases, i.e. if the types are only referred to by pointer or reference, we may be able to dodge the issue by replacing  `#include`s in DART headers with forward declarations. Many of these are also optional components - I'm not sure how we should handle that. @psigen may have some ideas.

I am very much looking forward to moving `osgDart` into the `dart::gui::osg` namespace and including it as a CMake component - the camel case name drives me crazy every time I clone the repository. :cry: 

@mxgrey When do you intend on making that change?

jeffeb3, 2016-03-08:
Optional component should be treated just like osgDart. If you want to use them, then add that component to the the single find_package and you should also get the extra dependencies.

mkoval, 2016-03-08:
I agree, but implementing that is tricky. I don't see any way to handle this without manually maintaining a list of transitive dependencies *for each optional component* in the `DARTConfig.cmake` file. Do you see a better way to handle this?

jeffeb3, 2016-03-08:
Each optional component would need their own dependencies, yes. That's not really more work, the components already have those dependencies. It starts as as simple as something like the grep you did. If two optional components depend on the same thing, then you can have the dependency on both. But if you really want stability, there should be a test suite for each rational combination of optional packages, and that's where it can be a time suck.

So obviously, it pays to be diligent ahead of time about which parts are optional. 

But the benefits are also obvious, if many users don't need urdf support, then they don't have to install it.

jslee02, 2019-03-13:
I believe this is resolved since now we can include `gui-osg` component by:
```cmake
find_package(DART REQUIRED COMPONENTS gui-osg)
# ...
target_link_libraries(<your_prj> ${DART_LIBRARIES})  # using variable or
target_link_libraries(<your_prj> dart-gui-osg)  # using target
```

closed 2019-03-13 by jslee02