dartsim/dart #620: Difference between velocity and servo ActuatorType

mkoval, 2016-02-23:
I am trying to simulate a velocity-controlled body pushing a position-controlled body. I attempted to dot his by setting the pusher `Skeleton`'s root `FreeJoint` to have an `ActuatorType` of `VELOCITY`. Surprisingly, the simulator seemed to completely ignore collisions between that `Skeleton` and other `Skeleton`s in the `World.

I changed the pusher's `ActuatorType` to `SERVO` and everything worked as expected. See [this video](https://youtu.be/nCU50bOf6JA) for a comparison of `VELOCITY` and `SERVO` with all else held constant.

I can't find anything in the documentation that indicates that this would not work:

> FORCE/PASSIVE/SERVO joints are dynamic joints while ACCELERATION/VELOCITY/LOCKED joints are kinematic joints.
> ...
> `SERVO`: Command input is desired velocity, and the output is joint acceleration. The constraint solver will try to track the desired velocity within the joint force limit. All the joint constraints are valid.
> ...
> `VELOCITY`: Command input is joint velocity, and the output is joint force. The joint velocity is always satisfied but it doesn't take the joint force limit into account. If you want to consider the joint force limit, should use SERVO instead. All the joint constraints are invalid.

It is likely that I'm missing some subtlety here because I am generally confused about `ActuatorType`:

- What does it mean for an actuator to "output" joint force or acceleration?
- What does it mean for an actuator to make "all constraints invalid"?
- What is the difference between a kinematic and dynamic joint for purposes of simulation?

@jslee02 Could you shine some light on this?

costashatz, 2016-09-23:
Any insight on this one?

I have one more example to illustrate. I have a very simple arm (4-dof) and I am using a very simple p-velocity controller to control it. When I use `VELOCITY` actuators everything works find, whereas with `SERVO` actuators I get:

```bash
*** Error in `./build/test': free(): invalid next size (fast): 0x0000000001a92250 ***
```

The code I am using to control the joints is:

```cpp
Eigen::VectorXd q = _robot->skeleton()->getPositions();
Eigen::VectorXd q_err = _target_positions - q;

double gain = 1.0 / (dart::math::constants<double>::pi() * _robot->skeleton()->getTimeStep());
Eigen::VectorXd vel = q_err * gain;

_robot->skeleton()->setCommands(vel);
```

Any ideas why this is not working?

costashatz, 2016-11-11:
Playing around with the servo actuators, for the same robot (simple 4-DOF arm) I am getting also other kind of errors:

```bash
 malloc.c:2373: sysmalloc: Assertion `(old_top == (((mbinptr) (((char *) &((av)->bins[((1) - 1) * 2])) - __builtin_offsetof (struct malloc_chunk, fd)))) && old_size == 0) || ((unsigned long) (old_size) >= (unsigned long)((((__builtin_offsetof (struct malloc_chunk, fd_nextsize))+((2 *(sizeof(size_t))) - 1)) & ~((2 *(sizeof(size_t))) - 1))) && ((old_top)->size & 0x1) && ((unsigned long) old_end & pagemask) == 0)' failed.
```

Tracking with gdb still leads back to:

```cpp
dSolveLCP(int, double*, double*, double*, double*, int, double*, double*, int*)
```

Any ideas on what's wrong?