dartsim/dart #622: Collisions with MeshShape showing strange behaviour

johnrowlay, 2016-02-26:
I have compiled release version 5.1 of DART, and run a simple demo with a MeshShape falling onto a BoxShape. However, this is causing very strange behaviour, with the MeshShape failing to come to rest, and oscillating indefinitely, whilst defying gravity. This same behaviour occurs with many meshes I have tried.

I know that meshes are only approximated by bounding boxes when it comes to computing volumes and inertia, but this seems to be more of a collision issue. Is this a bug, or are there special considerations I need to take into account when using meshes?

A video of the strange behaviour can be found here:
https://www.youtube.com/watch?v=5roCbUBK9wE&feature=youtu.be

My simple demo code that displays the above video can be found here:
http://freetexthost.com/6prqzwb0vv

And the mesh file used in the above can be found here:
https://www.dropbox.com/s/f2e2zu9ca94wqov/airplane.off?dl=0

mxgrey, 2016-02-26:
I've seen this behavior before. It's almost certainly because the inertial properties that you've given to the mesh (mass, center of mass location, and/or moment of inertia) do not match its geometry. DART is most likely doing the correct thing given the physical parameters that you've assigned the object; it's just that the physical properties you've given it (or that DART has given it by default) are not physically possible.

mxgrey, 2016-02-26:
To illustrate how this can happen, imagine this scenario:

You've constructed a mesh shaped like a cone, but you've specified that the center of mass of the shape is above the tip of the cone. Obviously this cannot really happen, but let's suppose we make a model like this and throw it in the simulator. If we drop the cone sideways, after it hits the ground it will end up balancing on the tip, even under the influence of perturbations. It will oscillate on the tip and appear to be defying gravity, even though it's doing the correct thing given the model that it was provided.

johnrowlay, 2016-02-26:
Yes, that has solved it -- thanks for the explanation. The meshes I was loading were not centred around (0, 0, 0), but after finding the local centre of mass from the mesh vertices (an approximation using a bounding box), and setting it for the body node, the behaviour is now much more reasonable.

closed 2016-02-26 by johnrowlay