dartsim/dart #625: Decrease minimum required bullet version from 2.82 to 2.81

jslee02, 2016-03-04:
I set the minimum required bullet version 2.82 in the first place without any investigation. This prevents us from using DART with bullet on trusty (bullet version of trusty is 2.81) unless we install higher bullet version manually. [According to the change log](https://github.com/bulletphysics/bullet3/blob/master/docs/BulletQuickstart.pdf), it seems there is no change in collision module of bullet in 2.82 so we would be able to use 2.81 without any trouble.

mkoval, 2016-03-04:
:+1: I also ran into this on Ubuntu 14.04

If 2.82 was chosen arbitrarily, is 2.81 any less arbitrary? Why not set the required version to just 2.0 unless we have a compelling reason to choose a later version?

jslee02, 2016-03-04:
It's not a compelling reason but the reason is that the supporting Ubuntu distribution is trusty, which provides bullet 2.81.

PR #636 (2016-03-20): Removed version check for Bullet.
This fixes #625 by removing the test for a Bullet version. This is important because Ubuntu 14.04 ships with Bullet 2.81, which fails the version check for Bullet 2.82 in the current DART `master`.

I would prefer to remove the check entirely, instead of simply lowering to Bullet 2.81, because we have no evidence to believe that the interface will not work on earlier versions of Bullet. We can re-introduce a version check if we find a compelling reason to do so (e.g. DART fails to build against very old versions).

I also removed the logic that uses `pkg-confi [... cut]
closed 2016-03-20 by jslee02