dartsim/dart #626: Ambiguous overload on MetaSkeleton::getLinearJacobianDeriv

mkoval, 2016-03-07:
Both of these overloads match `getLinearJacobianDeriv(const JacobianNode*)`:
```
  virtual math::LinearJacobian getLinearJacobianDeriv(
      const JacobianNode* _node,
      const Frame* _inCoordinatesOf = Frame::World()) const = 0;

  virtual math::LinearJacobian getLinearJacobianDeriv(
      const JacobianNode* _node,
      const Eigen::Vector3d& _localOffset = Eigen::Vector3d::Zero(),
      const Frame* _inCoordinatesOf = Frame::World()) const = 0;
```

Trying to call this overload produces a compile-time error. I suggest removing the default value on `_inCoordinatesOf` in the first overload to remove the ambiguity.

mxgrey, 2016-03-07:
Thanks for reporting this. I disagree with your suggested fix, though. The better fix would be to remove the default argument for ``_localOffset``, because the version that does not take in a local offset (the first one) is more efficient, so we should default to that one whenever possible.

I'm going to patch this up with a regression test right now.

jslee02, 2016-03-07:
I believe this is addressed by #628 .

closed 2016-03-07 by jslee02