dartsim/dart #627: Have a class for composing states and properties during inheritance

mxgrey, 2016-03-07:
We've established a ``State`` and ``Properties`` pattern which enables a huge number of features, but it comes at the cost of a significant amount of boilerplate code, even for very simple cases.

I propose we create a templated ``Compose`` class that can take care of this boilerplate work. Instead of inheriting a class directly, you would inherit it through the ``Compose`` class, and the ``Compose`` class would take care of defining the ``UniqueProperties`` and full ``Properties`` for your newly derived class.

A quick sketch up of the idea applied to Properties can be seen here: https://gist.github.com/mxgrey/ffd620c76f16ae5a5dd7

psigen, 2016-03-08:
Can you put together a gist where you define some simple example `State` and `Properties` using the current API and the proposed `Compose` API?  Maybe a toy problem like adding a `State` that holds a floating point value for each BodyNode?

I'm not sure I even fully understand the current API, so it's hard to figure out the change.

mkoval, 2016-03-08:
> I'm not sure I even fully understand the current API, so it's hard to figure out the change.

:+1:

mxgrey, 2018-05-04:
This has been finished and merged.

closed 2018-05-04 by mxgrey