dartsim/dart #629: Linking against ${DART_LIBRARIES} not working in Ubuntu 14.04

a-price, 2016-03-07:
Linking executable "test" against DART using the find_package supplied variable doesn't seem to be working for me:
`target_link_libraries(test ${DART_LIBRARIES})`
However, linking to "dart" works fine:
`target_link_libraries(test dart) `

Running the build system with `make VERBOSE=1` gives the following output using `${DART_LIBRARIES}`:

> c++    -std=c++0x    CMakeFiles/test.dir/test.cpp.o  -o test -rdynamic /usr/local/lib/libdart.so -Wl,-rpath,/usr/local/lib

but returns the following when using `dart`:

> c++    -std=c++0x    CMakeFiles/test.dir/test.cpp.o  -o test -rdynamic /usr/local/lib/libdart.so.5.1.1 /usr/local/lib/libdart-core.so.5.1.1 /opt/ros/indigo/lib/libccd.so /opt/ros/indigo/lib/libfcl.so -lassimp -lboost_regex -lboost_system -lGLU -lGL -lSM -lICE -lX11 -lXext -lglut -lXmu -lXi -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -lconsole_bridge -ltinyxml -ltinyxml2 -Wl,-rpath,/usr/local/lib:/opt/ros/indigo/lib 

So it would appear that the dependencies of `libdart.so` are not getting pulled in correctly in the first method. I've attached a minimum (non)working example:
[CMakeLists.txt](https://github.com/dartsim/dart/files/162159/CMakeLists.txt)
[test.txt](https://github.com/dartsim/dart/files/162164/test.txt)
(Apologies for the .txt extension for the .cpp file, GitHub doesn't support uploading .cpp's)

mkoval, 2016-03-07:
:+1: :+1: :+1: I've also ran into a lot of difficult-to-reproduce problems with `DART_LIBRARIES` missing transitive dependencies (e.g. `assimp`, `tinyxml`, etc).

It's important to remember that CMake has two different ways of `find_package`ing:

1. **module mode:** Searches for a `FindXXX.cmake` file in `CMAKE_MODULE_PATH` that *is provided by a third-party* for a package that does not use CMake as its build system. The `.cmake` file should use generate full paths using helper functions like `find_library` and `find_path`.

2. **config mode:** Searches for a `XXXConfig.cmake` file in `CMAKE_PREFIX_PATH` that is *generated and installed by the package's own `CMakeLists.txt` file. This `.cmake` file should include the auto-generated `XXXTargets.cmake` and, optionally, a `XXXVersion.cmake` files. These auto-generated files create imported CMake targets with the target properties required for linking (e.g. absolute path) already set.

DART currently mixes the two. It generates a `DARTTargets.cmake` file, but then resolves each library to an absolute path using `find_library`. This cause CMake to ignore all of the meta-data that `DARTTargets.cmake` file sets on the imported targets!

**TL;DR:** Fixing this should be as simple removing the `find_library` calls in `DARTConfig.cmake.in`.

jslee02, 2016-03-16:
I believe this is addressed by #630. Thanks @a-price !

closed 2016-03-16 by jslee02