dartsim/dart #642: FCLCollisionDetector -- multiple fcl::CollisionObjects refer to the same FCLCollisionNode?

mxgrey, 2016-03-21:
Running the latest ``master`` branch, I'm seeing a strange issue in ``testWorld`` and ``testVskParser`` where they are suddenly failing. It wasn't showing up until we started using [this line](https://github.com/dartsim/dart/blob/6f948a3352c8650da5a70d95ed4095fd5e49d53a/dart/constraint/ConstraintSolver.cpp#L70) by default, although I'm sure the issue has been lurking around for a long time as ``FCLCollisionDetector`` has gone unused.

As far as I can tell, the main issue seems to be that ``FCLCollisionDetector`` is mapping multiple ``fcl::CollisionObject``s to the same ``dart::collision::CollisionNode`` object and then attempting to check collisions for it. I have no idea what might be causing this.

@jslee02 is working on major changes to the whole collision library component, so maybe this will get addressed in his changes. In the meantime, should we perhaps revert to defaulting to ``FCLMeshCollisionDetector`` in order to avoid this issue?

jslee02, 2016-03-21:
Yeah, I also noticed this issue. I would like to revert the defaulting and address the `FCLCollisionDetector` issue in the work on collision library component to avoid doing work twice. I overlooked that `FCLCollsionDetector` is not stable yet.

jslee02, 2016-04-16:
I believe this is resolved by #631.

closed 2016-04-16 by jslee02