dartsim/dart #643: Proposal: Rename the class Addon to Component

mxgrey, 2016-03-23:
First, I want to apologize for revisiting an issue as contentious as naming. I believe I have some very compelling reasons for recommending this change, so I'm going to try to lay those reasons out as clearly as I can.

### What we currently have (on master branch):

We're using a ``State`` and ``Properties`` pattern to represent the parameters of kinematic and dynamic structures (as well as all other concrete objects in DART).

A ``State`` is a set of parameters which is expected to change arbitrarily often. Typically these are variables which are functions of time, although if a user creates a custom stateful object, they are given the liberty to decide which of their parameters fall under the umbrella of ``State``.

``Properties`` are parameters which are not expected to change very often. Ideally, a change to the ``Properties`` of an object will also increment the "version number" of that object, as well as the version number of any objects to which it is attached. Typically the data within a set of ``Properties`` are constant parameters pertaining to the equations of motion of an object, although if a user creates a custom object with custom properties, they are given the liberty to decide which of their parameters fall under the umbrella of ``Properties``.

The base objects of DART have their basic ``Properties`` data embedded in the class definition itself. For example, ``BodyNode::Properties`` has information about its name, inertia, and collidability. ``Skeleton::Properties`` has information about its name, timestep, and mobility. ``Joint::Properties`` has information about name, relative transforms, and actuator type. This information is stored internally in each of these classes as a member variable.

However, derived classes need to be able to extend this ``Properties`` information, and they need to add their own ``State`` information as well. They need to do it in a way that is directly accessible to the base class as well. Luckily, we've already invented the ``Addon`` class which was designed to allow users to arbitrarily attach custom classes that have their own ``State`` and ``Properties``. We can use this machinery to trivially make the ``States`` and ``Properties`` of derived classes accessible to their base class. Problem solved, right??

### Why is this a problem??

This scheme results in **two categories** of States and **two categories** of Properties. There are the basic ``State`` and ``Properties`` which are embedded in the base classes (like ``BodyNode``, ``Skeleton``, and ``Joint``) and then there are the extended ``State`` and ``Properties`` which are stored in the ``Addon``s (like ``SoftBodyAddon``, ``RevoluteJoint``, ``EulerJoint``, etc). This splits our State/Properties pattern along two separate lines, forcing us to implement machinery to merge these lines together in order to produce a **third category** of ``State`` and ``Properties``, which is the **complete** set of ``State``s and ``Properties``. [Here’s a link to illustrate this structure](https://gist.github.com/mxgrey/df0056ce66925105adeb).

At the moment, this extra machinery isn’t especially troublesome. It just means that we need set/get functions for this **third** type of State/Properties which then delegates to the setters/getters of the first and second types of State/Properties. However, this conceptual split is an artificial one which is resulting from our current implementation. It kind of derails the State/Properties pattern along two separate lines and then smashes them back together in a ham-fisted way.

This is problematic because it’s hard to predict what else we might want to do with our State/Properties pattern in the future, and having this kind of jagged edge could cause issues down the road.

### What is the proposed solution?

I propose we **eliminate** the need for this **third category** of State/Properties by including the embedded State/Properties with the ``Addon`` State/Properties. This way, the embedded State and Properties can be managed within the exact same machinery as the ``Addon`` State and Properties.

However, this poses a semantic issue: If these absolutely critical base State/Properties are being manipulated by a mandatory ``Addon``, can we really call it something that we’re “adding on” anymore? ``Addon`` would no longer represent an additional custom plugin, but instead they would be getting used for both critical embedded data and custom user data. This is what motivates my proposal to rename ``Addon`` to ``Component``.

### What would this entail as far as implementation goes?

Mostly this will just require a rename of the ``Addon`` and ``AddonManger`` classes. The machinery that exists is still perfectly suitable for our needs (although I’ll be implementing some specialized ``Addon``/``Component`` types that allow them to operate on data which is embedded in its Manager instead of in itself).

This will also reduce some of the existing machinery. [Here is what the earlier example would look like](https://gist.github.com/mxgrey/615416de55100d7eeb67) if we could eliminate the **third category** of State/Properties.

To summarize: This proposal would simplify our State/Properties implementation and produce a more unified and consistent pattern which would almost certainly be helpful as more features get built on top of DART, all with minimal changes to existing infrastructure (besides renaming a class).

Also, I’m open to names besides ``Component`` if someone has a better suggestion for this concept.

psigen, 2016-03-23:
Before we get into the weeds on this, thanks for making an excellent writeup of the issue @mxgrey!

psigen, 2016-03-23:
I am generally onboard with consolidating under the `Addon` system.  It will encourage us to standardize and clean up `Addon/AddonManager`, and it makes DART the internals less opaque.

As for the naming, I am a little hesitant to use `Component` as this is also the term that is used in our build process for each separable library.  Here is a list of other terms, perhaps people will see something they like:

* Attribute
* Element
* Module
* Aspect
* Characteristic
* Facet
* Attachment
* Extension
* Factor
* Fundamental
* Thing
* Part
* Piece
* Quality
* Atom
* Principle
* Ingredient

mxgrey, 2016-03-23:
I definitely see why ``Component `` is problematic. I like the sound of ``Aspect``. I'd say that's my favorite option from the list.

mkoval, 2016-03-23:
Thank you for the excellent explanation @mxgrey! I find the current implementation *very confusing* when inheritance is involved. For example, `RevoluteJoint` has the following machinery:

- Type `RevoluteJoint::Properties`
- Type `RevoluteJoint::UniqueProperties`
- Type `RevoluteJoint::AddOn`
- Type `Joint::Properties`
- Type `Joint::ExtendedProperties`
- Method `setProperties(RevoluteJoint::UniqueProperties)`
- Method `setProperties(RevoluteJoint::Properties)`
- Method `setProperties(Joint::Properties)`
- Method `setProperties(SingleDofJoint::UniqueProperties)`
- Method `setProperties(SingleDofJoint::Properties)`
- Method `RevoluteJoint::Properties getJointProperties()`
- Method `RevoluteJoint::Properties getRevoluteJointProperties()`
- Method `SingleDofJoint::Properties getSingleDofJointProperties()`

The `AddonManager`, `AddonManagerJoiner`, and `common::AddonManager::Properties` classes involved. I'm not sure how all of this plays together. Can you explain how the current system works with inheritance (and how `AddonManagerJoiner` is involved)? What would this look like in the new paradigm?

I am in favor of using the name `Component`. Your proposal is *exactly* an implementation of a [component-based design](https://en.wikipedia.org/wiki/Component-based_software_engineering). I see your point about people confusing `Component` confused with `COMPONENTS` in CMake, since we are using both, but I am not too worried about this. The two features are used in very different contexts and the names look very different (e.g. `dart-optimizer-nlopt` vs `DynamicsComponent`).

psigen, 2016-03-23:
> I am in favor of using the name Component. Your proposal is exactly an implementation of a component-based design.

I agree that this is a component-based design, but virtually _everything_ in modern object-oriented architecture falls under this definition (including the CMake usage we are conflicting with, which is _also_  exactly an implementation of a component-based design).  The page you linked refers to many other things which all exactly match this definition which are definitely not all named "component":

* Objects (e.g. OO-classes)
* Modules (in python, lua, etc.)
* Namespaces
* Dynamically-linked libraries
* JavaBeans
* RPC interface libraries

It's a really weak definition which covers a huge amount of good coding practice, I don't think it's a major driver for the naming convention we use.

mxgrey, 2016-03-23:
> Can you explain how the current system works with inheritance

My proposal doesn't *necessarily* change anything when it comes to the distinction of ``UniqueProperties`` versus ``Properties`` for derived classes, but I'm open to suggestions regarding that matter.

For example, a possible idea is to have a class called ``ComponentManager::Properties`` (or ``AspectManager::Properties`` depending on which name we go with) that would allow you to access the individual component properties based on type. For example, you could imagine something like: ``properties.get<RevoluteJoint::Component>();`` which would return a pointer to a ``RevoluteJoint::Component::Properties`` object if one is available, or else a ``nullptr`` if one is not available. The catch is that this would **require** all classes that are derivatives of ``Component`` to define a ``Properties`` object within its class scope, or else this function cannot be implemented. We do not currently impose this requirement for Addons. An Addon's properties can be completely invisible to the user and it will still work just fine behind the scenes. That said, I think this is a pretty reasonable constraint to place on ``Component``s, since invisible states are probably undesirable in any reasonable application. Ideally, if a particular Component derivative doesn't have any ``Properties`` at all, then we could have the ``get<T>()`` function always return a ``nullptr``. We could probably implement that using ``std::enable_if``. Here's an illustration of how I could see this working:

```
ComponentManager::Properties properties;
properties.getOrCreate<Joint::Component>()->mT_ParentBodyToChild = transform;
properties.getOrCreate<Joint::Component>()->mName = "some_joint";
properties.getOrCreate<RevoluteJoint::Component>()->mAxis = Eigen::Vector3d::UnitZ();

revoluteJoint->setProperties(properties);
```

Then we could get rid of the ``UniqueProperties`` versus ``Properties`` distinction and just rename all the ``UniqueProperties`` to ``Properties``. Then we use the ``ComponentManager::Properties`` concept to unify all the individual components' properties.

Of course everything I'm talking about for ``Properties`` would be applied to ``State`` as well.

All of these changes would be waaaay more extensive than what I originally proposed, but if they sound good, I'd be happy to implement it.

I also wonder if we could have a better name than ``ComponentManager``/``AspectManager``/etc. Maybe something like ``Composition`` (although I'm worried that might be too vague of a term).

mxgrey, 2016-03-25:
(Sorry for the delay, I've been busy running experiments this week)

@mkoval asked me a few questions outside of the conversation which I think are worth answering in here.

> Is all of the data actually stored in an AddOn somewhere? or is some of it still embedded in the Joint class?

In the current implementation, ``Joint::UniqueProperties`` are stored in the ``Joint`` class. This is information like name, transform to child, transform to parent, and actuator type. Due to a limitation with the current implementation of ``Addon``s, the derived properties such as ``SingleDofJoint::UniqueProperties``, ``MultiDofJoint<DOF>::UniqueProperties``, ``RevoluteJoint::UniqueProperties``, ``EulerJoint::UniqueProperties``, (etc) are each stored in an ``Addon``. I believe (currently without proof) that storing this information in ``Addon``s is actually causing some very slight performance reduction, because I've been noticing that ``testDynamics`` is taking about 10% longer to run than it used to for me.

The motivation for storing the data in ``Addon``s is to take advantage of the way ``Addon`` ``Properties`` are serializable. This allows us to have a ``Joint::Properties`` class which will trivially suck in the properties of all its derived classes in addition to any user-added Properties.

The proposal I have would replace the ``Addon`` concept with something similar but more unified. Currently I favor the term ``Aspect``. An ``Aspect`` would provide an API for accessing the ``State`` and ``Properties`` of some portion of an object, whether that portion is embedded in the object definition or something that has been appended by a user. This means that ``SingleDofJoint::UniqueProperties``, ``RevoluteJoint::UniqueProperties``, (etc) can go back to being embedded in the Joint object itself, and there will be a ``SingleDofJoint::Aspect`` and ``RevoluteJoint::Aspect`` that will allow the ``State`` and ``Properties`` of these derived classes to be serialized.

Some ``Aspect``s would grant access to data embedded in an object, but other ``Aspect``s would contain their own ``State`` and ``Properties`` data, independently of the object that they are attached to. This is more flexible than the current ``Addon`` paradigm.

> If I want to subclass a class with an AddOn, how do I do that?

For simple cases, you just inherit the class. But if your derived class needs to be specialized for its own particular ``Addon``s, then you need to use a templated ``Joiner`` class in order for all the specialized templates to be resolved correctly. You can inherit without using the templated ``Joiner`` class, but you'd be taking a risk of the specialization not working correctly. This same thing would be true of my ``Aspect`` proposal.

> Are all of the functions on Joint just wrappers that set attributes on an Addon stored internally?

Not for the base ``Joint`` class, but pretty much yes for all the derivative classes. This would no longer be the case in my ``Aspect`` proposal, though.

[Here is a rough sketch of what a Joint inheritance tree would look like for Properties using Aspects](https://gist.github.com/mxgrey/5dd018a549493d376b92). Note that ``Composite`` is the name that I've given to the class that would replace ``AddonManager``. The idea is that a ``Composite`` is a composition of aspects. The important things to notice are

1. ``Properties`` can be stored within the object itself, which keeps everything together tightly in memory, and that's good for the processor cache and for performance.

2. There are only two categories of ``Properties``: ``Aspect::Properties`` and ``Composite::Properties`` and they share a unified architecture. In the current implementation, we have three types of ``Properties``; two of the types have completely different architectures, and the third type exists to hammer the two architectures together in a gross way.

I'm going to begin an implementation of this tomorrow. I'm happy to answer any questions that anyone has in the meantime.

mklingen, 2016-03-30:
This is a great change because it makes components first-class citizens. It's more pure, less confusing. IMO even `Skeleton` should extend from component.

I suggest the name `Component` because its the standard for these kinds of frameworks. When I see "component" I know what it means immediately and when I see "aspect" I'm confused.

I've also suggested this before, but I don't know if it makes more sense. Instead of a `Composite`, just make `Component` recursive and store the components in a tree. Then you can create new components by simple recursive composition, and there is exactly one kind of `Property` and `State`, etc. Some components will just exist to add meta-component state or properties to the system.

mxgrey, 2016-04-14:
Sorry for the delay in my reply. I wanted to finish the implementation so that I'd having something concrete to point to before I addressed your concerns.

> I suggest the name Component because its the standard for these kinds of frameworks.

This is ironically the same reason that I want to avoid the name ``Component``: The Component Pattern is pretty well-known and well-established, but what we're doing isn't *exactly* the traditional Component Pattern, so I don't want to mislead people into thinking that it is. To my understanding (and please correct me if I'm wrong), the Component/Composite Pattern usually allows bunches of objects to be treated with the same interface as single objects.

While our pattern does allow this, that is not its primary purpose. The motivation behind the ``Aspect`` is to give some introspection to base classes and subclasses all the way up and down the inheritance hierarchy. The second post in #659 attempts to illustrate the purpose of this. What it boils down to is: There are many derived types, and each derived type has some arbitrary States and Properties that need to be accessible to their base class. The Aspects Pattern enables this introspection with little-to-no overhead. It also allows users to extend the functionality of DART classes by creating their own Aspects and appending those Aspects to object instances. The States and Properties of any appended Aspects will be handled seamlessly within the same framework as the Aspects which pertain to an object's inheritance hierarchy.

> This is a great change because it makes components first-class citizens.

Unfortunately, most Aspects will still be second-class citizens in some sense. There are many Aspects which only reflect the information of various levels within a single object's inheritance hierarchy. Some Aspects could be considered first-class citizens because they do more than just reflect some other object's information, but this tends to be the exception rather than the norm.

> Instead of a Composite, just make Component recursive and store the components in a tree.

This can definitely be done within the current framework. Anyone could design an Aspect which is, itself, a Composite. The State and Properties of that Aspect could then be its Composite::State and Composite::Properties. We do not currently offer a ``CompositeAspect`` out-of-the-box, but I could *very* easily implement one if there is any interest in that. The only catch is that this could entail a significant amount of dynamic allocation if there are too many levels of ``CompositeAspect``s stacked up on each other, but maybe I could avoid this if I implement it carefully.

##### All that being said:

I could still be convinced to change the name if enough people have strong enough preferences. The name ``Aspect`` has grown on me, and I think it does a better job of capturing the meaning of what it represents, but if people feel that this pattern is close enough to the traditional Composite Pattern and that ``Aspect``s deserve to be called ``Component``s, then I could change my mind.

But I think the best illustration for why I would prefer ``Aspect`` over ``Component`` would be to consider this question: Suppose we have a revolute joint. Is the axis of that joint a component of the joint or is it an aspect of the joint?

jslee02, 2016-05-09:
Could we close this as this is addressed by #659?

closed 2016-10-17 by mxgrey