dartsim/dart #656: Consider to using std::size_t instead of size_t

jslee02, 2016-04-04:
Basically, they are both same, but, to be pedantic, `size_t` is defined in `std` namespace. Using `std::size_t` would be the consistent way as we always (at least in the headers) specify `std` for stl types like `std::vector<...>` for instance.

mxgrey, 2016-04-04:
I am okay with pedantry :+1: .

mkoval, 2016-04-04:
Here's what the C++ standard says:

> **17.4.3.1.4 Types [lib.extern.types]**
>
> For each type `T` from the Standard C library, the types `::T` and `std::T` are reserved to the implementation and, when defined, `::T` shall be identical to `std::T`.
>
> These types are `clock_t`, `div_t`, `FILE`, `fpos_t`, `lconv`, `ldiv_t`, `mbstate_t`, `ptrdiff_t`, `sig_atomic_t`, `size_t`, `time_t`, `tm`, `va_list`, `wctrans_t`, `wctype_t`, and `wint_t`.

The `::T` and `std::T` types are defined in different header files:

>  [Example: The header `<cstdlib>` assuredly provides its declarations and definitions within the namespace `std`. It may also provide these names within the global namespace. The header `<stdlib.h>` assuredly provides the same declarations and definitions within the global namespace, much as in the C Standard. It may also provide these names within the namespace `std`. — end example ]

**TL;DR:** We should use `std::size_t` if we include `<cstddef>` and use `size_t` if we include `<stddef.h>`. If both headers are included, then we may use either.

mxgrey, 2016-04-04:
We've been preferring the C++ versions of headers to the exclusion of C versions, so that would imply that we should prefer ``std::size_t``. The catch is that it seems most STL implementations *do* provide ``size_t`` in the global namespace in ``<cstddef>``, so we *could probably* get away with continuing to use ``size_t`` even without ever using ``<stddef.h>``, but it wouldn't be strictly guaranteed to work.

Taking all that into consideration, I would favor the side of consistency, pedantry, and guarantees (``std::size_t``) over shortcuts (``size_t``).

PR #681 (2016-04-24): Use std::size_t instead of size_t
resolves #656

---
This change is [[image]](https://reviewable.io/reviews/dartsim/dart/681)
closed 2016-04-24 by jslee02