dartsim/dart #665: Automatically updating collision geometry when Shape is updated

jslee02, 2016-04-13:
The collision geometry of collision detection engine (e.g., `fcl::CollisionGeometry` and `btCollisionShape`) doesn't get updated when `dart::dynamics::Shape`'s properties are changed. It would be nice if the collision geometry would be updated whenever the Shape is updated. This could be achieved by adding `Signal` to `Shape`.

mxgrey, 2016-04-13:
This is largely related to #664 , although I could imagine that we may want to solve them with different methods.

jslee02, 2016-04-13:
Yeah, I thought to add this to #664 as a comment but didn't. Because I would prefer to resolve them as separate pull requests for easier review process.

jslee02, 2016-04-13:
Oh, this is a duplicate of #197. Closing.

closed 2016-04-13 by jslee02