dartsim/dart #671: Duplicate entries in Skeleton::mBodyNodes causing segfault in destructor

chapulina, 2016-04-15:
On [Skeleton::Init](https://github.com/dartsim/dart/blob/v4.3.5/dart/dynamics/Skeleton.cpp#L434), the vector of body nodes within the skeleton is rearranged with breadth first search. During the rearrangement, there are circumstances in which the same node gets added twice. For example, if the same body node is the child of two different joints.

This causes a segfault on the skeleton's [destructor](https://github.com/dartsim/dart/blob/v4.3.5/dart/dynamics/Skeleton.cpp#L87), where it tries to delete the same pointer twice.

I was able to avoid the crash with the following patch, but I'm not sure if this would break other things.

```diff
diff --git a/dart/dynamics/Skeleton.cpp b/dart/dynamics/Skeleton.cpp
index 9e8eb3f..3f6d0be 100644
--- a/dart/dynamics/Skeleton.cpp
+++ b/dart/dynamics/Skeleton.cpp
@@ -442,7 +442,8 @@ void Skeleton::init(double _timeStep, const Eigen::Vector3d& _gravity)
     {
       BodyNode* itBodyNode = queue.front();
       queue.pop();
-      mBodyNodes.push_back(itBodyNode);
+      if (std::find(mBodyNodes.begin(), mBodyNodes.end(), itBodyNode) == mBodyNodes.end())
+        mBodyNodes.push_back(itBodyNode);
       for (size_t j = 0; j < itBodyNode->getNumChildBodyNodes(); ++j)
         queue.push(itBodyNode->getChildBodyNode(j));
     }
```

The issue was discovered while debugging a test failure in Gazebo where the skeleton's destructor is called, see this [pull request](https://bitbucket.org/osrf/gazebo/pull-requests/2263/clean-up-entities/diff).

chapulina, 2016-04-15:
@scpeters

jslee02, 2016-04-15:
Presumably, the test tries to parse a model (maybe `pr2.world`?) that includes kinematic closed loop. It should be handled as a dynamic constraint, but as reported [here](https://bitbucket.org/osrf/gazebo/issues/913/dart-does-not-detect-closed-loop), DART currently doesn't support automatic closed loop constraint construction yet.

This patch would resolve the segfault by ignoring the loops, but the test will possibly fail anyways since the closed loop constraint will be violated.

I think the test in Gazebo should be disabled until this is resolved in DART.

By the way, thanks for reminding us that DART is missing this feature.

mxgrey, 2016-04-15:
The ``Skeleton::init`` function no longer exists in versions >= 5.0, because Skeleton initialization is now handled automatically. We could patch version 4.3 if Gazebo users still need it.

This does raise a question in my mind, though: The only ways I could imagine this happening is:

1. Two ``BodyNode``s claim to have the same child ``BodyNode``. That would result in a non-tree structure, which is not supported by DART.

2. The same ``BodyNode`` has been added to the tree twice. This wouldn't necessarily break the dynamics algorithms, but it would probably be indicative of an error somewhere else.

Rather than silently handle and ignore this situation as proposed, I think we should patch it to print a warning to the user that they have provided a malformed Skeleton.

scpeters, 2016-04-15:
Yes, we discovered this issue while simulating the pr2 ([joint_screw.cc](https://bitbucket.org/osrf/gazebo/src/1d1e3a542af81670f43a120e1df7190592bc4c0f/test/integration/joint_screw.cc?fileviewer=file-view-default#joint_screw.cc-478)). That test is actually already disabled for dart, but it doesn't exit until after loading the pr2 world, and it seg-faults during the destructor. Due to memory management issues in gazebo (shared pointer reference loops), the destructors weren't being called until recently. For now, we can keep the test disabled, but make sure it doesn't even load the PR2 model.

mxgrey, 2016-04-15:
I think it's reasonable to patch DART v4.3 so that it prints a warning and dodges the segfault. The resulting tree just won't reflect the original intent of the model, because it will not close that loop. I'll make a PR for this.

mxgrey, 2016-04-16:
The patch has been merged in #672

closed 2016-04-16 by mxgrey