dartsim/dart #686: Support of Intel Compiler (icc/icpc)

Aneoshun, 2016-04-22:
Hello,
From what I see in the CMakeList, Dart does not support Intel's compiler (icc / icpc).
Do you plan to add the support of this compiler at some point?

FYI, I tried to quickly had the compiler in the CMakeList to pass the configure phase, but the compilation fails quite quickly ("default" destructors not found, for instance).

Many thanks in advance!

mxgrey, 2016-04-22:
The ``default`` keyword is a feature of C++11. If the compiler is failing on things like that, then it does not have C++11 compatibility. According to [this webpage](https://software.intel.com/en-us/articles/c0x-features-supported-by-intel-c-compiler) the latest versions of the Intel compiler should support C++11, so you might just need to update your compiler version.

jslee02, 2016-04-22:
>  you might just need to update your compiler version.

Or, you might need to set proper compiler flags depending on your OS and compiler.

> * On Windows: when using Intel C++ compiler with Visual Studio 2010* or 2012*, the C++11 features supported by Visual C++ 2010/2012 are enabled by default. Use "/Qstd=c++11" to turn on the support for all other cases.
* On Linux or Mac OS X: the C++11 features supported by gcc on the path are enabled by default. Use "-std=c++11" to turn on the support for all other cases.

Aneoshun, 2016-04-22:
Thank you for your very-fast reply.

I am using a quite recent version of icpc (probably the most recent one):
```
icpc -V
Intel(R) C++ Intel(R) 64 Compiler for applications running on Intel(R) 64, Version 16.0.0.109 Build 20150815
```
with the proper cpp11 flag (-std=c++11) 

The issues that I am reporting need probably more details:
The first Issue I encounter when compiling with icpc is this one: 
```
dart/utils/VskParser.cpp(788): internal error: assertion failed at: "shared/cfe/edgcpfe/folding.c", line 10757
      return std::pair<dynamics::WeldJoint*, dynamics::BodyNode*>(nullptr, 
```

It seems that the conversion from a WeldJoint* to a Joint* (return type) fails here.  
If I change WeldJoint* into Joint* ( tell me if I am doing wrong here), then I can compile the library. 

However, when I want to compile the unittests, I get a link error:
```
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::TaskSpaceRegion::~TaskSpaceRegion()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::JacobianTranspose::~JacobianTranspose()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::JacobianDLS::~JacobianDLS()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::HierarchicalIK::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::InverseKinematics::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::InverseKinematics::Constraint::~Constraint()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::HierarchicalIK::Constraint::~Constraint()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::JacobianTranspose::~JacobianTranspose()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::HierarchicalIK::Constraint::~Constraint()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::InverseKinematics::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::HierarchicalIK::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::Analytical::~Analytical()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::TaskSpaceRegion::~TaskSpaceRegion()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::HierarchicalIK::Constraint::~Constraint()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::Constraint::~Constraint()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::Analytical::~Analytical()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::InverseKinematics::Constraint::~Constraint()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::Constraint::~Constraint()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::HierarchicalIK::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::HierarchicalIK::Objective::~Objective()'
../lib/libdart.so.6.0.0: undefined reference to `dart::dynamics::InverseKinematics::JacobianDLS::~JacobianDLS()'
../lib/libdart.so.6.0.0: undefined reference to `non-virtual thunk to dart::dynamics::HierarchicalIK::Constraint::~Constraint()'
```

As the compilation phase passes, I think that the problem does not come from a cpp11 compatibility issue.

mxgrey, 2016-04-22:
Failing to convert a ``WeldJoint*`` to a ``Joint*`` strikes me as a nonsensical Internal Compiler Error. That particular inheritance hierarchy is a very simple one, and ``WeldJoint*`` to ``Joint*`` is an upcast, so it should be completely trivial. Of course, there's certainly nothing wrong with what you did (changing that line from a ``WeldJoint*`` to a ``Joint*``), but it still shouldn't be a necessary change. Also the fact that the error says:

> dart/utils/VskParser.cpp(788): **internal error:** assertion failed at:

I think it's explicitly saying that it's an ICE, and not an issue with the code base.

The linking error also strikes me as some kind of ICE, because it's only affecting classes with default-defined destructors. It's as if the compiler isn't bothering to give those destructors any definitions.

Aneoshun, 2016-04-22:
Thanks for your reply again.

I am not a big fan of ICC neither (I prefer Clang) but it's a requirement for running experiments on the Cluster in my university.

Moreover, I totally agree with you, these errors are...just weird and illogical...

FYI, if I comment all the destructors mentioned in the error, I manage to compile the unittests. But then, when I run, for example, the testSkelParser, I get a segfault.
```

../bin/tests/testSkelParser
[==========] Running 9 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 8 tests from SkelParser
[ RUN      ] SkelParser.DataStructure
[       OK ] SkelParser.DataStructure (1 ms)
[ RUN      ] SkelParser.EmptyWorld
[       OK ] SkelParser.EmptyWorld (1 ms)
[ RUN      ] SkelParser.SinglePendulum
Warning [SkelParser.cpp:1840] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
Segmentation fault (core dumped)

```

After digging a lot around this segfault, it seems that it comes from calls to methods with the same name but in different classes (a namespace resolution error?). To be honest, after reaching this point, I stop investigating.. too much nonsensical to me...

mxgrey, 2016-04-22:
> it's a requirement for running experiments on the Cluster in my university.

I see; that's unfortunate :disappointed: 

> when I run, for example, the testSkelParser, I get a segfault.

If the virtual destructors are being removed entirely, then I could imagine that potentially leading to segmentation faults if objects aren't getting deleted properly. Deleting a base class reference without a virtual destructor is undefined behavior, so there's no telling what might result. Given how wacky ICC appears to be, the undefined behavior might manifest itself as something that appears to be a namespace resolution failure (although I'm really just speculating).

It's a little more effort than just commenting them out, but you may be able to try changing

```
virtual ~ClassName() = default;
```

to 

```
inline virtual ~ClassName() { }
```

However, this may result in compilation errors in places where an object's destructor definition needs to be visible to the destructor. In those cases, you'll have no choice but to put the destructor definition in a ``.cpp`` file.

Aneoshun, 2016-04-25:
Thank all of you for your replies.

With the inline, empty destructor, I get the exact same result (but I didn't get any compilation trouble):
```
[==========] Running 9 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 8 tests from SkelParser
[ RUN      ] SkelParser.DataStructure
[       OK ] SkelParser.DataStructure (0 ms)
[ RUN      ] SkelParser.EmptyWorld
[       OK ] SkelParser.EmptyWorld (2 ms)
[ RUN      ] SkelParser.SinglePendulum
Warning [SkelParser.cpp:1840] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
Segmentation fault (core dumped)
```

I managed to pass this segfault by changing 3 lines in the dynamics/Entity.cpp:
```
diff --git a/dart/dynamics/Entity.cpp b/dart/dynamics/Entity.cpp
index 3bdd1d8..6f5439d 100644
--- a/dart/dynamics/Entity.cpp
+++ b/dart/dynamics/Entity.cpp
@@ -87,7 +87,7 @@ Entity::~Entity()
 void Entity::setProperties(const Properties& _properties)
 {
   // Set name
-  setName(_properties.mName);
+ Entity::setName(_properties.mName);
 }
 
 //==============================================================================
@@ -290,7 +290,7 @@ void Entity::changeParentFrame(Frame* _newParentFrame)
     if (it != mParentFrame->mChildEntities.end())
     {
       mParentFrame->mChildEntities.erase(it);
-      mParentFrame->processRemovedEntity(this);
+      mParentFrame->Frame::processRemovedEntity(this);
     }
   }
 
@@ -305,7 +305,7 @@ void Entity::changeParentFrame(Frame* _newParentFrame)
       mParentFrame->mChildEntities.insert(this);
       mParentFrame->processNewEntity(this);
     }
-    notifyTransformUpdate();
+   Entity::notifyTransformUpdate();
   }

```
This allows me to go the the next segfault:
```
[==========] Running 9 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 8 tests from SkelParser
[ RUN      ] SkelParser.DataStructure
[       OK ] SkelParser.DataStructure (1 ms)
[ RUN      ] SkelParser.EmptyWorld
[       OK ] SkelParser.EmptyWorld (1 ms)
[ RUN      ] SkelParser.SinglePendulum
Warning [SkelParser.cpp:1840] [SkelParser] <damping> tag is now an element under the <dynamics> tag. Please see (https://github.com/dartsim/dart/wiki/) for more details.
[       OK ] SkelParser.SinglePendulum (2 ms)
[ RUN      ] SkelParser.SerialChain
[       OK ] SkelParser.SerialChain (3 ms)
[ RUN      ] SkelParser.RigidAndSoftBodies
Segmentation fault (core dumped)
```
It looks like an issue in the namespace resolution, and it's likely that the new segfault originates from the same kind of issue.
Did you already encounter something similar?

mxgrey, 2016-04-25:
That sounds like ICC failing to comply with the ISO Standard and/or having a variety of internal bugs. You shouldn't have to explicitly declare the ``Entity::`` scope when you're already in the ``Entity::`` scope, or the ``Frame::`` scope when you're already in the ``Frame::`` scope.

In fact, these changes can be problematic, because ``Frame::processRemovedEntity`` is a virtual function which has been given a default definition, but ``BodyNode::processRemovedEntity`` overrides it. By explicitly calling ``Frame::processRemovedEntity`` you tell it to call the default definition, which prevents the virtual function from using the ``BodyNode::processRemovedEntity`` override, and that will almost certainly result in bugs for DART.

jslee02, 2017-11-12:
Closing due to lack of a response. Please re-open if you have further trouble.

closed 2017-11-12 by jslee02