dartsim/dart #695: Linking Problem with libdart.so on ubuntu 15.10

lucaskotw, 2016-04-24:
Dear all,

This is my first time to use DART sim and I have some problem while testing my own code.

I've successfully installed dart 5.1 on my ubuntu which installed on parallel desktop. I tested by running the tutorial binaries in dart source files, and they work fine.

However, when I try to build my own code in another dir, let's say ~/proj/dart-test/ with main.cpp as follows.
`#include <dart/dart.h>
int main()
{
  dart::dynamics::SkeletonPtr skel; // works fine
  dart::dynamics::SkeletonPtr skel = dart::dynamics::Skeleton::create("skeleton"); // failed
  return 0;
}`
I separate the compiling process from linking process with the following command.
**compile** `g++ -c main.cpp -Wall -msse2 -fPIC -std=c++11 -I/usr/include/eigen3 -I/usr/include/bullet --verbose`
**link** `g++ -o main main.o- -lboost_system -lBulletSoftBody -lBulletDynamics -lBulletCollision -lLinearMath -lassimp -ldart -Wl,--verbose`

And the compile process is fine, no error or warning. However, link process dump the following message, which shows that undefined reference to Skeleton::create (where the verbose options shows that I've link the libdart.so). What did I miss?

**Error message**
`attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crt1.o succeeded
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crt1.o
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crti.o succeeded
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crti.o
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/crtbegin.o succeeded
/usr/lib/gcc/x86_64-linux-gnu/5/crtbegin.o
attempt to open main.o succeeded
main.o
attempt to open /usr/local/lib/libboost_system.so failed
attempt to open /usr/local/lib/libboost_system.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libboost_system.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libboost_system.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_system.so succeeded
-lboost_system (/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_system.so)
attempt to open /usr/local/lib/libBulletSoftBody.so failed
attempt to open /usr/local/lib/libBulletSoftBody.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libBulletSoftBody.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libBulletSoftBody.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libBulletSoftBody.so succeeded
-lBulletSoftBody (/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libBulletSoftBody.so)
attempt to open /usr/local/lib/libBulletDynamics.so failed
attempt to open /usr/local/lib/libBulletDynamics.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libBulletDynamics.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libBulletDynamics.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libBulletDynamics.so succeeded
-lBulletDynamics (/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libBulletDynamics.so)
attempt to open /usr/local/lib/libBulletCollision.so failed
attempt to open /usr/local/lib/libBulletCollision.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libBulletCollision.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libBulletCollision.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libBulletCollision.so succeeded
-lBulletCollision (/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libBulletCollision.so)
attempt to open /usr/local/lib/libLinearMath.so failed
attempt to open /usr/local/lib/libLinearMath.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libLinearMath.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libLinearMath.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libLinearMath.so succeeded
-lLinearMath (/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libLinearMath.so)
attempt to open /usr/local/lib/libassimp.so failed
attempt to open /usr/local/lib/libassimp.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libassimp.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libassimp.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libassimp.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libassimp.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libassimp.so succeeded
-lassimp (/usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libassimp.so)
attempt to open /usr/local/lib/libdart.so succeeded
-ldart (/usr/local/lib/libdart.so)
attempt to open /usr/local/lib/libflann_cpp.so failed
[... 49 lines cut ...]
/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
attempt to open /usr/local/lib/libgcc_s.so failed
attempt to open /usr/local/lib/libgcc_s.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libgcc_s.so succeeded
-lgcc_s (/usr/lib/gcc/x86_64-linux-gnu/5/libgcc_s.so)
attempt to open /usr/local/lib/libgcc.so failed
attempt to open /usr/local/lib/libgcc.a failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libgcc.so failed
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/libgcc.a succeeded
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/crtend.o succeeded
/usr/lib/gcc/x86_64-linux-gnu/5/crtend.o
attempt to open /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crtn.o succeeded
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crtn.o
libpthread.so.0 needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libboost_system.so
found libpthread.so.0 at /lib/x86_64-linux-gnu/libpthread.so.0
libm.so.6 needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libLinearMath.so
found libm.so at /usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libm.so
libz.so.1 needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libassimp.so
found libz.so.1 at /lib/x86_64-linux-gnu/libz.so.1
ld-linux-x86-64.so.2 needed by /usr/lib/gcc/x86_64-linux-gnu/5/libstdc++.so
found ld-linux-x86-64.so.2 at /lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
main.o: In function `main':
main.cpp:(.text+0xee): undefined reference to `dart::dynamics::Skeleton::create(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status`

jslee02, 2016-04-24:
It seems the linker fails to find proper libraries, but I honestly don't know much about using compiler and linker. I instead use CMake to build things.

If you willing to use CMake for your project, then could you take a look at [this example](https://github.com/dartsim/dart-examples/tree/dart-5.1) and [this patch](https://github.com/dartsim/dart/pull/696) if it's useful to you?

lucaskotw, 2016-04-24:
Thank you @jslee02 

I've tried the example. I get the following error message.
```
lucas@ubuntu:~/proj/dart-examples$ cmake .
-- Looking for DART - found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lucas/proj/dart-examples
lucas@ubuntu:~/proj/dart-examples$ make -j4
Scanning dependencies of target HelloWorld
[100%] Building CXX object HelloWorld/CMakeFiles/HelloWorld.dir/Main.cpp.o
Linking CXX executable ../bin/HelloWorld
CMakeFiles/HelloWorld.dir/Main.cpp.o: In function `main':
Main.cpp:(.text+0x5a): undefined reference to `dart::simulation::World::World(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
Main.cpp:(.text+0x8b): undefined reference to `dart::collision::BulletCollisionDetector::BulletCollisionDetector()'
Main.cpp:(.text+0x97): undefined reference to `dart::simulation::World::getConstraintSolver() const'
Main.cpp:(.text+0xa2): undefined reference to `dart::constraint::ConstraintSolver::setCollisionDetector(dart::collision::CollisionDetector*)'
CMakeFiles/HelloWorld.dir/Main.cpp.o: In function `__static_initialization_and_destruction_0(int, int)':
Main.cpp:(.text+0x17d): undefined reference to `boost::system::generic_category()'
Main.cpp:(.text+0x189): undefined reference to `boost::system::generic_category()'
Main.cpp:(.text+0x195): undefined reference to `boost::system::system_category()'
CMakeFiles/HelloWorld.dir/Main.cpp.o: In function `btAlignedAllocator<int, 16u>::deallocate(int*)':
Main.cpp:(.text._ZN18btAlignedAllocatorIiLj16EE10deallocateEPi[_ZN18btAlignedAllocatorIiLj16EE10deallocateEPi]+0x18): undefined reference to `btAlignedFreeInternal(void*)'
CMakeFiles/HelloWorld.dir/Main.cpp.o: In function `btAlignedAllocator<btHashInt, 16u>::deallocate(btHashInt*)':
Main.cpp:(.text._ZN18btAlignedAllocatorI9btHashIntLj16EE10deallocateEPS0_[_ZN18btAlignedAllocatorI9btHashIntLj16EE10deallocateEPS0_]+0x18): undefined reference to `btAlignedFreeInternal(void*)'
CMakeFiles/HelloWorld.dir/Main.cpp.o: In function `btAlignedAllocator<btTriangleInfo, 16u>::deallocate(btTriangleInfo*)':
Main.cpp:(.text._ZN18btAlignedAllocatorI14btTriangleInfoLj16EE10deallocateEPS0_[_ZN18btAlignedAllocatorI14btTriangleInfoLj16EE10deallocateEPS0_]+0x18): undefined reference to `btAlignedFreeInternal(void*)'
collect2: error: ld returned 1 exit status
HelloWorld/CMakeFiles/HelloWorld.dir/build.make:86: recipe for target 'bin/HelloWorld' failed
make[2]: *** [bin/HelloWorld] Error 1
CMakeFiles/Makefile2:75: recipe for target 'HelloWorld/CMakeFiles/HelloWorld.dir/all' failed
make[1]: *** [HelloWorld/CMakeFiles/HelloWorld.dir/all] Error 2
Makefile:75: recipe for target 'all' failed
make: *** [all] Error 2
```

So maybe the problems are 

1. my compiler couldn't even find boost and bullet library?
2. some settings are missed because I've run ubuntu on parallel desktop?

It seems that the error comes from the setting of the compiler

P.S. I get the similar undefined reference even on Mac (OSX El Captain)

Is there any settings I should add? Or there's other problem?

jslee02, 2016-04-24:
Quick question: Did you install DART 5.1 with this [patch](https://github.com/dartsim/dart/pull/696)?

lucaskotw, 2016-04-24:
Silly question: How could I install the patch? Should I remove first, download, and install it again?

jslee02, 2016-04-24:
If you [installed DART from the dart ppa](https://github.com/dartsim/dart/wiki/DART%205.1%20Installation%20for%20Ubuntu#install-dart-using-ubuntu-packages), then you should remove it first as:
```shell
sudo apt-get remove libdart5-dev
sudo apt-get remove libdart-core5-dev
```

Or, if you [installed DART from the source](https://github.com/dartsim/dart/wiki/DART%205.1%20Installation%20for%20Ubuntu#install-dart-from-source), then you don't need to remove it. It will be overwritten when you install a new one.

A quick way to test [the example]() would be:
* Remove `libdart5-dev` and `libdart-core5-dev` if installed
* [Download DART 5.1 that includes the patch](https://github.com/dartsim/dart/archive/js/bullet_include_dir/5.1.zip)
* Build and install the downloaded DART following [this instruction](https://github.com/dartsim/dart/wiki/DART%205.1%20Installation%20for%20Ubuntu#install-dart-from-source)
* Build the example

lucaskotw, 2016-04-24:
Thank you for you zip with patch.

However, I run into series of error of nplot and osgDart support while building DART `make -j4`
(I installed the DART from source instead of ppa)
```
[  0%] Built target gtest
[  1%] Built target gtest_main
[ 46%] Built target dart-core
[ 47%] [ 47%] Built target dart-optimizer-ipopt
Building CXX object dart/optimizer/nlopt/CMakeFiles/dart-optimizer-nlopt.dir/NloptSolver.cpp.o
[ 58%] Built target dart
[ 66%] Built target osgDart
Scanning dependencies of target testBuilding
Scanning dependencies of target testCommon
Scanning dependencies of target testCollision
[ 66%] Building CXX object unittests/CMakeFiles/testCommon.dir/testCommon.cpp.o
[ 66%] [ 67%] Building CXX object unittests/CMakeFiles/testCollision.dir/testCollision.cpp.o
Building CXX object unittests/CMakeFiles/testBuilding.dir/testBuilding.cpp.o
In file included from /home/lucas/proj/dart-js-bullet_include_dir-5.1/dart/optimizer/nlopt/NloptSolver.h:40:0,
                 from /home/lucas/proj/dart-js-bullet_include_dir-5.1/dart/optimizer/nlopt
... (remove similar error)
/usr/include/nlopt.hpp:399:10: note:   conversion of argument 1 would be ill-formed:
/home/lucas/proj/dart-js-bullet_include_dir-5.1/dart/optimizer/nlopt/NloptSolver.cpp:153:61: error: invalid conversion from ‘double (*)(unsigned int, const double*, double*, void*)’ to ‘nlopt::vfunc {aka double (*)(const std::vector<double>&, std::vector<double>&, void*)}’ [-fpermissive]
                                       mProperties.mTolerance);
                                                             ^
Building CXX object unittests/CMakeFiles/testCompositeResourceRetriever.dir/testCompositeResourceRetriever.cpp.o
dart/optimizer/nlopt/CMakeFiles/dart-optimizer-nlopt.dir/build.make:54: recipe for target 'dart/optimizer/nlopt/CMakeFiles/dart-optimizer-nlopt.dir/NloptSolver.cpp.o' failed
make[2]: *** [dart/optimizer/nlopt/CMakeFiles/dart-optimizer-nlopt.dir/NloptSolver.cpp.o] Error 1
CMakeFiles/Makefile2:428: recipe for target 'dart/optimizer/nlopt/CMakeFiles/dart-optimizer-nlopt.dir/all' failed
make[1]: *** [dart/optimizer/nlopt/CMakeFiles/dart-optimizer-nlopt.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX executable ../bin/tests/testBuilding
[ 68%] Built target testBuilding
Linking CXX executable ../bin/tests/testCompositeResourceRetriever
[ 68%] Built target testCompositeResourceRetriever
Linking CXX executable ../bin/tests/testCollision
[ 68%] Built target testCollision
Makefile:126: recipe for target 'all' failed
make: *** [all] Error 2
```

Should I remove the optional support for DART first?

lucaskotw, 2016-04-24:
I solved it by reboot the virtual machine to make sure there's enough swap space. And I built DART with the patch again. The patch works fine and the example you gave runs well.

Thank you very much!

jslee02, 2016-04-25:
Great! Once #698 is merged, you can use the patched version of DART from the PPA back.

closed 2016-04-25 by jslee02