dartsim/dart #705: Understanding FreeJoint

mklingen, 2016-04-27:
So as I understand it, `FreeJoint` is a 6-DOF joint allowing for translation and rotation. There are of course many ways of representing translation and rotation. But I see some mixed forms are used in `FreeJoint`. The dofs seem to be defined as `[ rot_x, rot_y, rot_z, pos_x, pos_y, pos_z,]`, meaning position and Euler angles, right? But some other parts of the documentation imply that the joint is using "Generalized Coordinates" (like, plucker coordinates? screw coordinates?), and in other places it seems to be using `Isometry3d`, which is a 4x4 homogenous representation.

So my question is, what is the underlying representation returned by `getPositions`? Is it valid for me to call `setPositions(positions + delta)` with `delta` being a 6D vector representing position and rotation change? Or should I be passing in deltas in the tangent space?

mklingen, 2016-04-27:
To clarify, all I need right now are two functions:

    // Add a small 6DOF vector delta to the joint's position
    Pose retract(Vector6 delta)
    {
        // ?
        return joint->getPositions() + delta;
    }

   // This is the inverse of "retract". Given a second set of coordinates,
   // what is the 6DOF delta required?
   Vector6 localCoordinates(Pose other)
   {
       // ?
       return delta - joint->getPositions();
   }

Usually for a 6DOF transformation I would do this using the tangent space doing something like this:

    Pose retract(Vector6 delta)
    {
        return expmap(delta) * pose;
    }

   Vector6 localCoordinates(Pose other)
   {
       return logmap(pose.inverse() * other);
   }

mxgrey, 2016-04-28:
The names that are assigned to the DOFs are deceptive. Each name is only accurate when the other two DOFs are set to zero. ``FreeJoint`` uses expmap, so you need to use the tangent space. However, there is a static function:

```
static Eigen::Vector6d FreeJoint::convertToPositions(const Eigen::Isometry3d&)
```

which will convert a transform into a suitable 6D vector. The exact meaning of the ``FreeJoint`` degrees of freedom are subject to change (they have changed at least once in the past), but this static function is guaranteed to do the right thing for you.

There is also a function called ``Joint::integratePositions(double dt)`` which can be used for the kind of delta operations that you're talking about, but there's a catch: The function is state-dependent. It uses the current velocity data and integrates that velocity, but scaled by ``dt``. The result is then stored in the Joint's position data.

We have raised issue #699 to create the kind of function that you're talking about. Most likely, that feature will fall out of the geometric joint implementation that @jslee02 is working on right now.

mklingen, 2016-04-29:
Okay thank you. This is very helpful. I've decided to just control my code so that 6D vectors don't get used directly but are inferred from the poses, which I modify in the tangent space. Then I apply these to the joint using helper functions.

closed 2016-04-29 by mklingen