dartsim/dart #710: Consider hosting multiple versions of API documentation

jslee02, 2016-04-28:


jslee02, 2016-10-15:
This could be done using multiple GitHub repositories. [GitHub Pages](https://pages.github.com/) allows us to host webpage per repository. The html files could be in master branch or gh-pages branch. For example, I created [jslee02/v6.0.1](https://github.com/jslee02/v6.1.0) and put the API html files generated by doxygen at the master branch, and [this](https://jslee02.github.io/v6.1.0/) is the website of it. We could have multiple dartsim/[VERSION] repositories in similar way.

One thing I'm not sure is which version of API documentation should be maintained. We could host:
(1) master, most recent versions of each major release like: 6.1.1, 5.1.4, and 4.3.6
(2) master, most recent versions of each minor release like: 6.0.1, 6.1.1, 5.0.x, and so on

I think (1) would be a reasonable option since we intend to keep the API compatibility within the same major version.

psigen, 2016-10-15:
I don't know if multiple repos is the right way to handle this. We would end up with hundreds of them over time, which seems wildly unmanageable.

What is the downside of building docs into subdirectories of a single `gh-pages` branch? The normal branches of the repo could be individual doc versions, with the special `gh-pages` branch having a built version of each per subdirectory? This update could be done as a post automated build step.

jslee02, 2016-10-15:
Hm, I haven't tested if subdirectories work. If so, it sounds better solution. I'll test now.

jslee02, 2016-10-15:
It works perfectly! Thanks @psigen .

Okay then we could host as many versions as we want.

I think we could use master branch for having the subdirectory instead of gh-pages since the repo is only for API hosting but not for other source or something like [this](https://github.com/jslee02/v6.1.0/tree/master).

psigen, 2016-10-15:
Sure ` gh-pages` is just the default, but master makes sense if it's a pure
documentation repository.

jslee02, 2018-02-13:
I believe this issue is resolved. (see: http://dartsim.github.io/dart/).

closed 2018-02-13 by jslee02