dartsim/dart #717: BulletCollisionDetector doesn't distinguish two groups

jslee02, 2016-05-08:
Collision checking for two collision groups should the test pairs of two objects from different groups rather than from one group. However, it was not straightforward to implement `BulletCollisionDetector` to work so since I couldn't find a way of using bullet's collision masks or filter callback after the collision objects are added to `btCollisionWorld` to update the collision pairs.

mkoval, 2016-05-09:
I am not sure how to interpret this bug. I see two possibilities about `BulletCollisionDetector`:

1. group-group checks are less efficient than they could be, but return the correct result
2. group-group checks return the incorrect result

Which of these is true?

jslee02, 2016-05-09:
Both. 

> group-group checks are less efficient than they could be, but return the correct result

It couldn't be efficient as `fcl` is since bullet doesn't support checking a group against to another group by default. We should create another group and put all the collision objects into it from the two groups for this.

> group-group checks return the incorrect result

To make this to work correctly, we should set the proper collision mask when the collision objects are added `btCollisionWorld`, but this is not possible with the current API of DART. I just saved this for the next release.

mkoval, 2016-05-10:
I am still not sure what the current behavior is. What is the outcome if I attempt to perform a group-group check with Bullet in the current `master` version of DART?

jslee02, 2016-05-10:
Suppose we have object1 in group1, and also have object2 and object3 in group2 where only object2 and object3 are colliding. In this case, we would expect group1-group2 returns false because the testing pairs should be object1-object2 and object1-object3 (objects in group1 against object in group2). However, BulletCollisionDetector would have all the objects in one group, which is temporarily created in collide() function, and check all the possible pairs including object2-object3. So it would return true, which is not correct behavior.

mkoval, 2016-05-10:
I see. That is a pretty major bug. 😰 

If we can't easily fix this in the 6.0.0 release, perhaps we can add a warning to `collide(CollisionGroup, CollisionGroup)` on `BulletCollisionDetector`?

jslee02, 2021-07-02:
Fixed in 6.11.0 by #1585

PR #1585 (2021-07-02): Fix incorrect group-group collision checking for BulletCollisionDetector
Fix #717

***

**Before creating a pull request**

- [x] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [x] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change
closed 2021-07-02 by jslee02