dartsim/dart #729: Dart 5.1 simwindow non consistent frame rate, glutTimerFunc is called more than once.

JiaYeLi, 2016-05-21:
If I press "space key" twice, the frame time rate will increase.
And simulation window display discontinuous results.

so that I add some codes in MyWindow::keyboard() near where "glutTimerFunc" is called.

//initilization firstPressPlay = true
case ' ':
      mSimulating = !mSimulating;
      if (mSimulating) {
        mPlay = false;
        if(firstPressPlay)   //added line
            glutTimerFunc(mDisplayTimeout, refreshTimer, 0);
        firstPressPlay = false;   //added line
      }
      break;

Is there a better way to solve this problem?

tianyang9310, 2016-07-27:
I have come across the same problem. Is there any official fix?

jslee02, 2016-10-14:
This should be resolved by #794.

jslee02, 2016-10-15:
Closing. Reopen if it still doesn't work.

closed 2016-10-15 by jslee02