dartsim/dart #730: Ways to load and parse ASF/AMC motion data

lucaskotw, 2016-05-24:
Is there an elegant way to load [ASF/AMC motion data](http://research.cs.wisc.edu/graphics/Courses/cs-838-1999/Jeff/ASF-AMC.html) and apply them to a skeleton? I just simply parse the data and assign the kinematic data. It will be great if there's a built-in method.

If there has no handy ways to read them, could I propose a way to parse it? ASF/AMC format seem a neat way to record the motion capture data.

jslee02, 2016-05-24:
:+1: It sounds good to me. I don't know about the format, but it looks like VSK file. VskParser and other parser classes would be helpful when you implement the parser. Just let us know if you need any help.

lucaskotw, 2016-05-25:
Thanks! I'll try to implement first and post related issues.

jslee02, 2021-07-12:
Closing this issue for now. Feel free to reopen when there is an update :)

closed 2021-07-12 by jslee02