dartsim/dart #733: Proposal: Support mimic joint

jslee02, 2016-06-06:
Reference: http://wiki.ros.org/urdf/XML/joint

costashatz, 2018-10-19:
@jslee02 what is the status on this one? I need this for one of my projects and since I will spend some time providing this functionality, it might be worth it doing it in a generic way for the whole project. Where do you think this is best to add? In the joint class or as a constraint? To me it sounds better if we add it as a constraint (something similar to the servo motor constraint). This means that we should also add a new type of joint (mimic joint).

In short, the mimic joint should try to follow the position of the "parent" joint and we can impose force/torque constraints in a similar fashion to the servo motor; i.e., assume that the joint is velocity controlled, compute the max velocity to accomplish the desired position (taking into account the time-step) and then enforce max force/torque. What do you think?

I will try to have a sketch later today or coming from Monday..

jslee02, 2018-10-19:
It sounds great! No progress from my end yet.

Off the top of my head, I would treat "mimic joint" as a new [actuator type](https://github.com/dartsim/dart/blob/master/dart/dynamics/Joint.hpp#L65-L71) instead of a new type of joint. The new actuator type would work similar to `SERVO` as you mentioned. For this, we might need a place to store the "parent" (or "master") joint and need to update the target velocity of the "mimic joint" automatically. A minimal change would be adding it to `Joint.hpp`. This would be redundant for other actuator types though.

Let me update here if I have a better idea.

costashatz, 2018-10-22:
> Off the top of my head, I would treat "mimic joint" as a new [actuator type](https://github.com/dartsim/dart/blob/master/dart/dynamics/Joint.hpp#L65-L71) instead of a new type of joint.

That's what I meant; a new actuator type..

jslee02, 2018-10-25:
Resolved by #1178

closed 2018-10-25 by jslee02