dartsim/dart #736: Use upstream fclConfig.cmake once FCL 0.5 is released

jslee02, 2016-06-25:
The current FCL provides `fclConfig.cmake` by [#116](https://github.com/flexible-collision-library/fcl/pull/116), and it will probably be included in version 0.5. We could use it instead of our [`FindFCL.cmake`](https://github.com/dartsim/dart/blob/master/cmake/FindFCL.cmake) to find fcl in CMake. The only required change would be replacing `find_package(FCL REQUIRED)` to `find_package(fcl REQUIRED)`.

jsbruglie, 2018-03-07:
Will DART support FCL 0.5 in the near future?
I am currently working with [MoveIt!], which uses FCL 0.5, thus preventing me from having it installed alongside DART.

(The ultimate objective is using ROS + Gazebo 9.0 + Dart 6 + MoveIt!)

[MoveIt!]: http://moveit.ros.org/

jslee02, 2018-03-07:
DART already support [FCL 0.5](https://github.com/dartsim/dart/pull/749) since 6.0.1 and [FCL 6.0](https://github.com/dartsim/dart/pull/936) (current master branch) since 6.4.0.

What is exactly preventing you from using DART and MoveIt now?

jsbruglie, 2018-03-07:
Thank you! I wrongly assumed DART required `libfcl-dev` debian package, which conflicts with `libfcl-0.5-dev`
I already managed to get the required setup by building FCL from source as well and tweaking the default installation directories. Thank you once more!

mxgrey, 2018-03-08:
After poking around, one straightforward solution for us to consider would be to add DART to the ROS distribution ecosystem. That way, people who want to use DART in ROS can easily do so without any conflicts.

The ROS-distributed package would be named according to the distribution that it belongs to, e.g. `ros-kinetic-dart`. We already provide a catkin `package.xml` for DART, so it should be pretty straightforward to make this happen. The full instructions can be found [here](http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease).

I'll begin work on making this happen, if no one has any objections. I'll start from version 6.3.0, since that appears to be our latest tagged release. I don't imagine there's much point in backporting to older versions of DART.

jslee02, 2018-03-08:
That sounds exciting! I have some questions though.

> The ROS-distributed package would be named according to the distribution that it belongs to, e.g. ros-kinetic-dart, so it wouldn't conflict with the libdart packages in our own PPA.

Just to clarify, `ros-kinetic-dart` (or `ros-kinetic-dartsim` as we're planning to change the package name to `dartsim`) and `libdart*` couldn't be installed alongside, right?

I'm not sure if it is a good practice to have two packages of ROS package and non-ROS package. I thought `libfcl-0.5-dev` is a stopgap solution for that `libfcl-dev` in Trusty is too old. Wouldn't this possibly cause other version conflicts? I mean in the future ROS + Gazebo 9.0 + DART 6 + MoveIt! wouldn't be possible anyway unless we have `ros-kinetic-gazebo`, which is the reverse direction of Gazebo had gone through. Am I missing something here?

mxgrey, 2018-03-08:
They can be installed alongside each other because `ros-kinetic-dart` will be installed to `/opt/ros/kinetic/` instead of `/usr/`. For an end user to even "see" `ros-kinetic-dart` during their build configuration, they would need to run the `/opt/ros/kinetic/setup.bash` file. At that point, cmake rules take over to decide whether to use the system installation or the `/opt/ros/kinetic` installation (although `catkin` might have some precedence rules of its own that I don't know about).

> I thought libfcl-0.5-dev is a stopgap solution for that libfcl-dev in Trusty is too old

I don't know the original motivation behind `libfcl-0.5-dev`. It's not clear to me whether it was intended for general consumption or whether it was only intended for ROS-related packages. If it were only intended for ROS-related packages, we could have avoided this conflict by making a `ros-kinetic-fcl` package instead of `libfcl-0.5-dev`, because the `ros-kinetic` version would go into `/opt/ros/kinetic`.

Since that isn't the case, I can only assume that `libfcl-0.5-dev` was intended for broader use, and thus it gets installed to the system. This is unfortunate since it seems to belong to an Open Robotics PPA, which makes it difficult for other packages in other PPAs to depend on. At the same time, it conflicts with the debian `libfcl-dev` package, so... it's a bit of a mess :/

> I mean in the future ROS + Gazebo 9.0 + DART 6 + MoveIt! wouldn't be possible anyway unless we have `ros-kinetic-gazebo`

A user can have gazebo-9 installed to their system, and this combination would work just fine. This combo also works fine if someone installs DART strictly by building it from source. The only issue is that the deb package for `MoveIt!` depends on `libfcl-0.5-dev` while the deb package for `libdart6-dev` depends on `libfcl-dev`, but there is an explicit conflict between `libfcl-0.5-dev`'s deb and `libfcl-dev`'s deb, so we cannot simultaneously install the `MoveIt!` deb and the `libdart6-dev` deb.

However, since DART has no issue with building against `libfcl-0.5-dev`, we can build DART from source without any trouble. It's just that the DART deb doesn't give us the flexibility to choose between `libfcl-dev` or `libfcl-0.5-dev`; we have to pick one or the other, as far as I can tell, but we can't pick `libfcl-0.5-dev` without requiring users to expose themselves to the ROS PPA (... not that the ROS PPA would be bad for users, but it's an extra step that shouldn't be necessary in a perfect world).

Also, gazebo-9 doesn't depend on FCL at all (at least from what I can tell), so gazebo-9 is completely independent of this whole issue.

Truthfully, the only thing we're really "accomplishing" by putting DART into the ROS ecosystem is:
1. We create an alternative deb package for DART which is able to depend on `libfcl-0.5-dev` instead of `libfcl-dev`.
2. ROS users have an easier time accessing DART because they can get it through the Open Robotics PPA instead of our dartsim PPA.

> (or ros-kinetic-dartsim as we're planning to change the package name to dartsim)

I wasn't totally clear on whether that decision was final, so thanks for mentioning it. I'm on board with using `dartsim`, so I'll be sure to use that when referring to the package name from now on.

jslee02, 2018-03-08:
Thanks for the clear explanation! Let me know if there is anything I can help for having DART in the ROS ecosystem.

> I wasn't totally clear on whether that decision was final, so thanks for mentioning it.

Yeah, maybe I didn't mention that I'm also on board with that. :stuck_out_tongue:

mxgrey, 2018-04-23:
I thought I'd mention here that I recently noticed `dartsim` is now available in kinetic under the package `ros-kinetic-dartsim`. This should make it much much easier for ROS users to take advantage of DART inside ROS projects, alongside MoveIt!.

The one catch is that it's still not possible to install both debian packages `libdart6-dev` and `ros-kinetic-dartsim` from their apt repos simultaneously, because of the conflict between the `libfcl-dev` and `libfcl-0.5-dev`.

That being said, we could avoid this specific conflict if we were more strict about separating interface and implementation. We have a complete abstraction layer around `fcl`, so in principle we shouldn't need `libfcl-dev`. The problem right now is we're exposing some implementation headers for the FCL collision detector. I don't exactly know if it's worth the effort, but it's something to consider.

PR #1640 (2021-12-24): Use fclConfig.cmake
resolves #736 

***

#### Before creating a pull request

- [ ] Document new methods and classes
- [ ] Format new code files using ClangFormat by running `make format`
- [ ] Build with `-DDART_TREAT_WARNINGS_AS_ERRORS=ON` and resolve all the compile warnings

#### Before merging a pull request

- [ ] Set version target by selecting a milestone on the right side
- [ ] Summarize this change in `CHANGELOG.md`
- [ ] Add unit test(s) for this change
closed 2021-12-24 by jslee02