dartsim/dart #751: dominoes tutorial doesn't work on mac

AidanRocke, 2016-07-22:
I get 20 errors:

/usr/local/include/dart/common/detail/Cloneable.hpp:128:55: error: a space is

      required between consecutive right angle brackets (use '> >')

  return common::make_unique<MakeCloneable<Base, Mixin>>(*this);

                                                      ^

/usr/local/include/dart/common/detail/Cloneable.hpp:212:1: error: 'auto' not

      allowed in function return type

auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(

^~~~

/usr/local/include/dart/common/detail/Cloneable.hpp:213:22: error: expected ';'

      at end of declaration

    const Data& data) -> ProxyCloneable&

                     ^

/usr/local/include/dart/common/detail/Cloneable.hpp:213:23: error: cannot use

      arrow operator on a type

    const Data& data) -> ProxyCloneable&

                      ^

/usr/local/include/dart/common/detail/Cloneable.hpp:223:1: error: 'auto' not

      allowed in function return type

auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(

^~~~

/usr/local/include/dart/common/detail/Cloneable.hpp:224:17: error: expected ';'

      at end of declaration

    Data&& data) -> ProxyCloneable&

                ^

/usr/local/include/dart/common/detail/Cloneable.hpp:224:18: error: cannot use

      arrow operator on a type

    Data&& data) -> ProxyCloneable&

                 ^

/usr/local/include/dart/common/detail/Cloneable.hpp:234:1: error: 'auto' not

      allowed in function return type

auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(

^~~~

/usr/local/include/dart/common/detail/Cloneable.hpp:235:33: error: expected ';'

      at end of declaration

    const ProxyCloneable& other) -> ProxyCloneable&

                                ^

/usr/local/include/dart/common/detail/Cloneable.hpp:235:34: error: cannot use

      arrow operator on a type

    const ProxyCloneable& other) -> ProxyCloneable&

                                 ^

/usr/local/include/dart/common/detail/Cloneable.hpp:245:1: error: 'auto' not

      allowed in function return type

auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(

^~~~

/usr/local/include/dart/common/detail/Cloneable.hpp:246:28: error: expected ';'

      at end of declaration

    ProxyCloneable&& other) -> ProxyCloneable&

                           ^

/usr/local/include/dart/common/detail/Cloneable.hpp:246:29: error: cannot use

      arrow operator on a type

    ProxyCloneable&& other) -> ProxyCloneable&

                            ^

/usr/local/include/dart/common/detail/Cloneable.hpp:308:1: error: 'auto' not

      allowed in function return type

auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::get() const -> Data

^~~~

/usr/local/include/dart/common/detail/Cloneable.hpp:308:72: error: expected ';'

      at end of declaration

auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::get() const -> Data

                                                                       ^

/usr/local/include/dart/common/detail/Cloneable.hpp:308:73: error: cannot use

      arrow operator on a type

auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::get() const -> Data

                                                                        ^

In file included from tutorialDominoes.cpp:32:

In file included from /usr/local/include/dart/dart.hpp:33:

In file included from /usr/local/include/dart/common/common.hpp:4:

In file included from /usr/local/include/dart/common/AspectWithVersion.hpp:35:

/usr/local/include/dart/common/detail/AspectWithVersion.hpp:57:3: error: 

      unknown type name 'constexpr'

  constexpr static void (*UpdateState)(Derived*) = updateState;

  ^

/usr/local/include/dart/common/detail/AspectWithVersion.hpp:57:13: error: 

      expected member name or ';' after declaration specifiers

  constexpr static void (*UpdateState)(Derived*) = updateState;

  ~~~~~~~~~ ^

/usr/local/include/dart/common/detail/AspectWithVersion.hpp:113:3: error: 

      unknown type name 'constexpr'

  constexpr static void (*UpdateProperties)(Derived*) = updateProperties;

  ^

fatal error: too many errors emitted, stopping now [-ferror-limit=]

20 errors generated.

mkoval, 2016-07-22:
It looks like your compiler does not support C++11, which is required by DART (or we're missing the `--std=c++11` flag somewhere). What version of OS X are you using and compiler are you using?

AidanRocke, 2016-07-22:
I'm using OS X El Capitan and the gcc compiler.

AidanRocke, 2016-07-22:
Ok. I just tried:

`Aidans-MacBook-Pro:tutorials aidanrocke$ gcc -std=c++11 -o tutorialDominoes tutorialDominoes.cpp
In file included from tutorialDominoes.cpp:32:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:6:
In file included from /usr/local/include/dart/common/Composite.hpp:35:
/usr/local/include/dart/common/detail/CompositeData.hpp:35:10: fatal error: 
      'Eigen/Core' file not found
#include <Eigen/Core>
         ^
1 error generated.`

What is Eigen/Core?

psigen, 2016-07-22:
`Eigen` is a [matrix library for c++](http://eigen.tuxfamily.org/index.php?title=Main_Page) and one of the [dependencies of DART](https://github.com/dartsim/dart/wiki/DART%206.0%20Installation%20for%20Mac#install-required-dependencies)

I think some of your confusion is coming from the tutorials omitting details on how they should be built.  Like most C++ libraries, DART is very tricky to manually build from a command-line compiler.  Usually, you would use a build system like CMake to manage the many flags for things like including libraries and setting up directory paths.

If you have the DART source code checked out, you should be able to build tutorials using the included `CMakeLists.txt` build files.

https://github.com/dartsim/dart/wiki/DART%206.0%20Installation%20for%20Mac#build-and-install-dart

jslee02, 2016-07-22:
If you would like to use DART as an external library, [here](https://github.com/dartsim/dart-examples) is a simplest example.

AidanRocke, 2016-07-26:
Ok. This time I'm getting:

`Aidans-MacBook-Pro:tutorials aidanrocke$ gcc -std=c++11 -o tutorialDominoes-Finished
clang: error: no input files`

AidanRocke, 2016-07-26:
Ok. This time I'm getting:

Aidans-MacBook-Pro:tutorials aidanrocke$ gcc -std=c++11 -o tutorialDominoes-Finished clang: error: no input files

ffnc1020, 2017-08-25:
Just installed dart on mac, I always get this error message.

`/usr/local/include/dart/config.hpp:5:10: fatal error: 'Eigen/Core' file not found`

Got same error using Xcode to compile.

mxgrey, 2018-04-07:
It seems like the ability to build examples from outside the DART source tree has been added. Feel free to reopen the issue if you continue to have issues.

closed 2018-04-07 by mxgrey