dartsim/dart #753: Build error with Visual Studio 2015 Update 3

iamyaoting, 2016-07-22:
Hello，I  install the Visual Studio Update3, but I built it with following error:
`2>D:\Projects\Cpp\3rdsource\dart\dart/common/AspectWithVersion.hpp(80): error C2974: 'dart::common::detail::AspectWithState': invalid template argument for 'CompositeT', type expected (compiling source file D:\Projects\Cpp\3rdsource\dart\dart\collision\CollisionGroup.cpp)
2>  D:\Projects\Cpp\3rdsource\dart\dart/common/detail/AspectWithVersion.hpp(48): note: see declaration of 'dart::common::detail::AspectWithState' (compiling source file D:\Projects\Cpp\3rdsource\dart\dart\collision\CollisionGroup.cpp)
2>  D:\Projects\Cpp\3rdsource\dart\dart/dynamics/EndEffector.hpp(57): note: see reference to class template instantiation 'dart::common::AspectWithStateAndVersionedProperties<dart::dynamics::Support,dart::dynamics::detail::SupportStateData,dart::dynamics::detail::SupportPropertiesData,dart::dynamics::EndEffector,dart::dynamics::detail::SupportUpdate,dart::dynamics::detail::SupportUpdate>' being compiled (compiling source file D:\Projects\Cpp\3rdsource\dart\dart\collision\CollisionGroup.cpp)`
What's the problem? Can you help me?
Thank you

jslee02, 2016-07-22:
The support for Visual Studio 2015 building DART 6.0 has been suspended because of the incomplete `constexpr` support of VS2015. However, it seems VS2015 Update 3 has resolved the issue, and there are other build issues as you reported. I'll look into it when having a chance.

iamyaoting, 2016-07-22:
Ok, As you say, in update 3, I did not see the constexpr error. 
But the template errors came out. 
Thank you~

mottosso, 2016-12-04:
I can confirm that the same error occurs with Visual Studio 2017 RC1.

```
C:\Users\marcus\dart\dart\common\AspectWithVersion.hpp(80): error C2974: 'dart::common::detail::AspectWithState': invalid template argument for 'CompositeT', type expected (compiling source file C:\Users\marcus\dart\dart\simulation\Recording.cpp) 
```

Along with these.

```
C:\Users\marcus\dart\dart\dynamics\detail\EntityNode.hpp(57): error C2228: left of '.mName' must have class/struct/union (compiling source file C:\Users\marcus\dart\dart\dynamics\SimpleFrame.cpp) 
C:\Users\marcus\dart\dart\dynamics\EntityNode.hpp(65): error C3668: 'dart::dynamics::EntityNode<dart::common::RequiresAspect<dart::common::EmbeddedPropertiesAspect<DerivedT,PropertiesDataT>>>::setName': method with override specifier 'override' did not override any base class methods 
C:\Users\marcus\dart\dart\dynamics\EntityNode.hpp(68): error C3668: 'dart::dynamics::EntityNode<dart::common::RequiresAspect<dart::common::EmbeddedPropertiesAspect<DerivedT,PropertiesDataT>>>::getName': method with override specifier 'override' did not override any base class methods 
```

jslee02, 2016-12-04:
Hm, I'm not sure whether these are bugs of VS or our code has syntactic errors.

marijnfs, 2017-01-17:
Any progress on this issue? Since there are no prebuilt libraries, it seems very constrictive to require a such an old VS for building

jslee02, 2017-01-19:
Sorry, no progress yet. I'm still getting the same errors with Visual Studio 2017 RC1 as @mottosso .

marijnfs, 2017-01-20:
For now I compiled the old release (<3) which works. It's annoying though, it seems it's just this one weird template inference problem. There is probably an easy fix, but the strange way templates are used here are a mystery to me.

jslee02, 2017-03-08:
Update: Visual Studio 2017 is now [available](https://www.visualstudio.com/downloads/), and AppVeyor seems to support the image for it [soon](https://github.com/appveyor/ci/issues/1179#issuecomment-285126349). We could test DART 6.1 on AppVeyor with VS2017 + Clang toolchain at that moment.

mgexo, 2018-05-30:
The original template problem of the author iamyaoting  still persists in even in very latest VS2017 version 15.7.2 and git master branch ( DART 7.0.0 ) in code that is supposed to use a cmake compiled version of dart ( also DART 6.6.0 had the same problem ):
```

1>dart\common\aspectwithversion.hpp(81): error C2974: 'dart::common::detail::AspectWithState': invalid template argument for 'CompositeT', type expected
1>dart\common\detail\aspectwithversion.hpp(50): note: see declaration of 'dart::common::detail::AspectWithState'
1>dart\dynamics\endeffector.hpp(143): note: see reference to class template instantiation 'dart::common::AspectWithStateAndVersionedProperties<dart::dynamics::Support,dart::dynamics::detail::SupportStateData,dart::dynamics::detail::SupportPropertiesData,dart::dynamics::EndEffector,dart::dynamics::detail::SupportUpdate,dart::dynamics::detail::SupportUpdate>' being compiled

```

Also switching VS2017 to "ISO C++ Latest Draft Standard (/std:c++latest)" does not help.

How exactly (with what version ) did you test that with VS2017 before closing ?
It happens simply when I include this file in code that wants to use dart:
`
#include <dart/simulation/World.hpp>
`

Also the dart library shows a huge number of warnings in VS2017 , so I need to use the following:

```
#pragma warning(disable:4250)
#pragma warning(disable:4244)
#pragma warning(disable:4996)
```

mxgrey, 2018-05-30:
I think the flag you need in your consumer project is [`/permissive-`](https://blogs.msdn.microsoft.com/vcblog/2016/11/16/permissive-switch/).

We should probably make that flag a `target_compile_options( ... PUBLIC ...)` flag of the DART library so that consumer projects will automatically get it.

Let us know if adding `/permissive-` to your project doesn't help.

mxgrey, 2018-05-30:
Note that the `/std:c++...` flag seems to be related to toggling the availability of standard *features* whereas the `/permissive-` flag actually affects the way that code gets parsed.

mgexo, 2018-05-30:
Indeed you are right, changing in VS2017 the "Conformance mode" in C++/Language to "Yes (/permissive-) " solves that issues. 

Thank you very much for the fast answer!