dartsim/dart #754: Missing tag for the 6.0.1 release

mkoval, 2016-07-22:
There are tags for all of patch version releases except for 6.0.1. I think this may have been a simple oversight, e.g. forgot to push the tag after creating it.

jslee02, 2016-07-22:
Actually, we haven't released yet pending #741.

mkoval, 2016-07-22:
I have `v6.0.1` installed from the DART PPA on Ubuntu 14.04, so it's definitely released:
```
$ dpkg-query --show libdart6
libdart6    6.0.1.ppa4~trusty
```

jslee02, 2016-07-22:
I messed up the procedure of releasing 6.0.1. `6.0.1.ppa4~trusty` was for testing. I should've used different package name like `libdart6-prerelease`.

Anyways, I will finish the procedure in this order:

1. Merge #741.
2. (probably) Upload `libdart6-prerelease` for testing.
3. Tag v6.0.1.

mkoval, 2016-07-22:
I don't think that is a good idea. It's not possible to re-release a version of a Debian package because it will cause everyone who previously `apt-get update` to get a hash mismatch. It also means that some people (like me) will have a version `6.0.1` installed that does not match the `v6.0.1` tag.

Instead, I suggest that you tag the version of the code that you did release as `v6.0.1` and change the release in #741 to 6.0.2. 😓

jslee02, 2016-07-22:
> Instead, I suggest that you tag the version of the code that you did release as v6.0.1 and change the release in #741 to 6.0.2.

That sounds good to me.

jslee02, 2016-07-22:
[v6.0.1](https://github.com/dartsim/dart/releases/tag/v6.0.1) is created.

closed 2016-07-22 by jslee02