dartsim/dart #758: FCL & inertia values set to zero cause unexpected behaviour

bpapaspyros, 2016-08-23:
Hello,

I am using DART with the FCL collision detector for simulating a humanoid robot and I am facing the following problem:

When loading a urdf file that contains links that act as **frames** (meaning we set mass and inertia values to zero, in an effort to ignore those parts that are useful for different purposes) or regular links containing zero inertia values the FCL crashes or some of the robot's links move unnaturally even when not in contact with objects/ground (arms wobbling, for lack of better term,  or head moving left and right even when it is set to stay explicitly in its position). Changing the values from zero to something smaller (e.g. 0.01). When trying to use even smaller values (e.g. 0.000001) unusual/bizarre behaviours resurface. 

I suppose the small/zero values cause numerical stability issues. Do you have any suggestions about that or have any ideas of something I could be setting wrong ? The simulation is fairly simple, meaning that I load the urdf and try to move just the legs and the arms (4 joints). The urdf in question can be found [here](https://github.com/robotology-playground/icub-model-generator/blob/master/generated/robots/iCubNancy01/model.urdf) . And a couple of suspicious links can be spotted [here](https://github.com/robotology-playground/icub-model-generator/blob/master/generated/robots/iCubNancy01/model.urdf#L400) and [here](https://github.com/robotology-playground/icub-model-generator/blob/master/generated/robots/iCubNancy01/model.urdf#L769).

- This could be related to #436 

I apologize for the vague description, but I am still trying to pinpoint the source of the problem and FCL is the first component that seems to be affected. Let me know if you need more details.

Thank you

mxgrey, 2016-08-23:
I tend to view this as a "garbage in -> garbage out" scenario. Zero-inertia links are physically "garbage", so we would expect any simulation results that use them to be "garbage".

I can think of two ways to handle your particular situation:

### Option 1: Make it sane

If you give some arbitrary non-zero (but small, so that it doesn't affect the rest of the robot) inertia properties to your bodies, you can avoid the worst of the numerical instability. Then you can use the ``Joint::LOCKED`` actuator mode to keep the head still. This should be the easiest option, as long as you have access to the DART API (but I'm not sure if this is possible with Gazebo).

### Option 2: Use visual shapes

Instead of making a ``link`` with a ``joint`` for your neck, you can add the neck geometry as a ``visual`` to the ``chest`` link with an offset transform that will place the neck geometry in the same place that it would have originally. The neck geometry will still exist within a "frame", which will be accessible like this:

```
ShapeNode* neck = skeleton->getBodyNode("chest")->getShapeNode(n);
```

where ``n`` is the index that the neck geometry ends up in.

This would allow the neck geometry to remain inertialess, collisionless, and fixed in place without producing garbage output. You will also continue to get the frame semantics that are available to bodies, because ``ShapeNode`` is a ``Frame`` in the same way that ``BodyNode`` is. This should also be completely possible with Gazebo. The catch is that accessing the neck frames might be awkward.

bpapaspyros, 2016-08-23:
Option 1, seems like the way to go (not using Gazebo) and confirms my suspicions. Option 2 is very interesting as well. Thank you !

closed 2016-09-27 by jslee02