dartsim/dart #762: Static+concave and other mesh types in Bullet collision detector

morphiccrab, 2016-08-26:
This addresses release-5.1 and 6.1. Tested on 5.1. In 6.1 there is an unrelated bug in the Bullet collision detector in the current HEAD so I didn't test it there, but looking at the sourcecode the issue is similar.

When using the Bullet collision detector, the default Bullet mesh type is a btConvexTriangleMeshShape. This obviously only supports convex meshes. For concave meshes that are additionally static, this type should be changed to btBvhTriangleMeshShape (in `dart/collision/bullet/BulletCollisionNode.cpp` and `dart/collision/bullet/BulletCollisionNode.h`. The constructor takes two extra booleans: `new btBvhTriangleMeshShape(btMesh, true, true)`). The collision detection then works correctly for meshes that have a weld constraint to the world. Bullet also supports other types of meshes (like GImpactShape) which should be supported.

How should this issue be handled? Should there be an extra XML tag defined in the SKEL specifiction that sets the collision handler for a given (mesh) skeleton?

jslee02, 2016-08-26:
Thanks for the reporting. I would prefer switching `btConvexTriangleMeshShape` to `btBvhTriangleMeshShape` rather than introducing an extra XML tag for convex shape. We could then add it when we actually have a shape type for convex meshes.

jslee02, 2016-08-26:
I think the simplest solution would be switching `btConvexTriangleMeshShape` of DART 5.1 to `btGImpactMeshShape` (not `btConvexTriangleMeshShape`) because [`btGImpactMeshShape` is only used for fixed object](http://bulletphysics.org/Bullet/BulletFull/classbtBvhTriangleMeshShape.html).

Could you post what was the unrelated issue of Bullet collision detector in DART 6.1?

jslee02, 2016-10-15:
I believe this is resolved by #764. Reopen if it still doesn't work for you.

closed 2016-10-15 by jslee02