dartsim/dart #767: Discussion: Consider using Eigen::AffineCompact3d for transformation representations

jslee02, 2016-08-28:
`Eigen::Isometry3d` is used everywhere in DART for the representation of transformations, but I'm considering replacing it with `Eigen::AffineCompact3d` for the following two reasons:

* `AffineCompact3d` consumes less memory than `Isometry3d` by not storing the last row of the transformation matrix, which is always `[ 0 0 0 1 ]`. This allows `AffineCompact3d` doesn't need to initialize the last row while `Isometry3d` requires the initialization.

* The default inverse operator of `Isometry3d` is the isometry inversion (e.g., `[ R^T -R^T*p; 0 1]`), but `AffineCompact3d` also can do this by passing a flag as `tf.inverse(Eigen::Isometry)`.

This is not an urgent issue, but I'd like to hear other's opinions and to know if I'm missing other advantages of using `Isometry3d`.

a-price, 2017-02-19:
I've noticed that `Eigen::Affine3d` is used pretty extensively in the ROS/MoveIt world, so I'm sure there's a compelling benefit for someone. My issue is code like this:

    Eigen::Affine3d T = Eigen::Affine3d::Identity();
    Eigen::Quaterniond q(T.rotation());

looks pretty harmless on initial inspection, but the `T.rotation()` call actually uses SVD to solve for the rotation, since the `R` matrix could have affine scale/shear components (even though this one does not). `Eigen::AffineCompact3d` does the same.

I tried using something like this on a planner a couple years ago, and was confused when the performance went through the floor.

jslee02, 2018-07-25:
@a-price Thanks for your insight on this!

> the T.rotation() call actually uses SVD to solve for the rotation, since the R matrix could have affine scale/shear components (even though this one does not). Eigen::AffineCompact3d does the same.

I wish it allows us to pass a hint that says the rotation matrix is orthornomal so it doesn't perform SVD unnecessarily, but there is no.

Closing this issue because I don't think the less memory consumption outweigh the disadvantages pointed out above.

closed 2018-07-25 by jslee02