dartsim/dart #771: (#757) FCL0.5 FCLCollisionNode boost undefined for tag 'v5.1.2'

jturner65, 2016-09-26:
#757 Is still a problem for tag v5.1.2 . Folks on Macs are getting : 

dart/dart/collision/fcl/FCLCollisionNode.cpp:354:5: error: use of undeclared identifier 'boost'

jslee02, 2016-09-26:
Could you use v5.1.3? v5.1.3 is the patched version for the issue. v5.1.3 hasn't released yet, but you can still build it from source.

jturner65, 2016-09-26:
I didn't see a tag v5.1.3.  Is there one?  This is for cs7496 and so I'd like to be able to give the students an easy solution if possible.

----- Reply message -----

https://github.com/dartsim/dart/issues/771#issuecomment-249602731

jslee02, 2016-09-26:
One quick solution you could try would be creating a new branch for CS7496 based on v5.1.3 and cherry pick the changes of #757.

closed 2016-10-17 by jslee02