dartsim/dart #780: Trouble with dart::dynamics::Inertia and unaligned arrays

JenniferBuehler, 2016-10-05:
Hi,

I'm getting [Eigen's unaligned array assertion failure](http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html) on a very simple program (linking to dart 6.1 debug library, with release library it just segfaults):

```cpp
#include <iostream>
#include <dart/dynamics/Inertia.hpp>

int main (int argc, char**argv)
{
  std::cout<<"one"<<std::endl;
  dart::dynamics::Inertia * inr = new dart::dynamics::Inertia();
  std::cout<<"two"<<std::endl;
  delete inr;
  std::cout<<"three"<<std::endl;
  return 0;
}
```

Output:
```
one
test: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 36; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Aborted (core dumped)
```

Inertia already defines ``EIGEN_MAKE_ALIGNED_OPERATOR_NEW`` so not sure why this still happens... Ideas?

JenniferBuehler, 2016-10-05:
Building with DART_ENABLE_SIMD=false fixes this. So not urgent for now but we should maybe look into this, I'll try later if I have the same on my other computer as well.

jslee02, 2016-10-09:
Tested on trusty x64, xenial x64, and xenial x86, but couldn't reproduce it. What was your configuration?

JenniferBuehler, 2016-10-11:
It was xenial x64. I had DART_ENABLE_SIMD=ON and was using the debug build.

I will test it on other systems as well and see if I get the same there.

jslee02, 2016-10-12:
I believe this has something to do with #790. One quick test to see if this is the case would be building the above code with `-march=native` option with DART built on the same machine.

JenniferBuehler, 2016-10-12:
Thanks for the hint!
I'll be checking back in on this issue on Friday, before I'm busy. I will let you know what the outcome is.

JenniferBuehler, 2016-10-14:
Alright I'm testing on a new machine now, and I get the same assertion failure.

I'm using xenial again. Output of ``uname -a``:

``4.4.0-43-generic #63-Ubuntu SMP <DATE> x86_64 x86_64 x86_64 GNU/Linux``

Building with ``-march=native`` works. So seems it really is related to #790. How about adding a cmake flag DART_DEFINITIONS to the DARTConfig.cmake, as an easy solution?

jslee02, 2016-10-14:
> How about adding a cmake flag DART_DEFINITIONS to the DARTConfig.cmake, as an easy solution?

Yeah, I thought that too at first. It is however still dangerous. `march=native` is machine dependent. If someone uses DART built on a different machine (e.g., from PPA or the ubuntu package server) of different CPU, the enabled SIMD instructions could be different. For this reason, `DART_ENABLE_SIMD` is disabled by default, and should be used only in the situation that DART and the consuming project are built on the same machine.

JenniferBuehler, 2016-10-14:
That makes sense. However, DART_ENABLE_SIMD was enabled by default both times I built it from scratch (one time master branch, other time v6.1 tag). I had to disable it. Or did you mean it is disabled by default for building the deb packages?

jslee02, 2016-10-14:
I meant it's disabled by #790 that is not in v6.1.0 nor master of  when you tried. #790 is now included in the current master and release-6.1 branch.

JenniferBuehler, 2016-10-14:
Ah, got it. Guess this is resolved then, feel free to close the issue.

Thanks for your help!

jslee02, 2016-10-14:
Great! DART 6.1.1 will be released soon probably with one another patch. Closing this issue.

closed 2016-10-14 by jslee02