dartsim/dart #792: pause/unpause causes redraw to act strangely

jturner65, 2016-10-14:
If I pause and then unpause a simulation, the frame count per refresh increases.  For example, if I run a simulation that shows the count increasing 1 frame per refresh (1->2->3->4 etc), if i pause and then unpause, this will change to 2 (4->6->8->10) and then 3 and so on.  This is also observable as a noticeable speedup in simulation rendering (students in cs7496 have remarked on it in their projects, on multiple platforms)

A reproducible example would be operational space control app. here's a video clip of the bug in action - note the change in delta mFrame displayed on the bottom of the screen.  Each time the frame count returns to 0 is from me pressing the space bar, and then it restarts when i press the space bar again.

https://dl.dropboxusercontent.com/u/55351229/frameCountBug.avi

jslee02, 2016-10-14:
This should be resolved by #794.

jturner65, 2016-10-15:
Thanks JS - which tag/branch should i use for this?

jslee02, 2016-10-15:
I guess you're using DART 5. In that case, you could use tag [`v5.1.4`](https://github.com/dartsim/dart/tree/v5.1.4) or branch [`release-5.1`](https://github.com/dartsim/dart/tree/release-5.1).

jturner65, 2016-10-15:
Yes, i need to tell the students in CS7496 which branch/tag to use - which would be better, that also addresses #757 ?

jslee02, 2016-10-15:
Using tag would be safer since branch gets changed over time. #757 should be addressed in the branch and tag.

jturner65, 2016-10-15:
Great, I'll check it out and then close this issue.  Thanks!

jturner65, 2016-10-15:
Unfortunately v5.1.4 is giving me CMAKE error : 

"CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FLANN_LIBRARIES
    linked by target "dart" in directory D:/dart5-1_cs7496/dart/dart"

I remember this was an issue we discussed in an email a little while ago, that using tag 5.1.2 addressed.

jslee02, 2016-10-15:
In a nutshell, that's not an issue of DART but the flann package installed by dart-prerequisites. It doesn't install necessary binary files.

A simple workaround is not using flann by commenting out [this line](https://github.com/dartsim/dart/blob/release-5.1/CMakeLists.txt#L253) since it's for planning component that I don't think it's required for the class. I would suggest creating an branch with the workaround for the class so that the students doesn't need to do this by their own.

jturner65, 2016-10-15:
commenting that line causes the flann not found branch of the subsequent if to execute.  should the entire section be commented?

would it be better to just update the v5.1.2 code with the single gluttimerfunc call in glutwindow?

jslee02, 2016-10-15:
I created a [branch](https://github.com/dartsim/dart/tree/cs7496-dart-5.1). I hope this works.

jturner65, 2016-10-15:
oh, great, i'll check it out.

jslee02, 2016-10-17:
I believe the issue of this post is resolved by #794. Let's open a separate issue for `flann` if necessary.

closed 2016-10-17 by jslee02