dartsim/dart #793: Trouble with DART on Ubuntu 16.04

pchokanson, 2016-10-14:
I am having some trouble running any of the examples, and have reduced at least one error to the following line of code (which should be trivial).

```
#include "dart/dart.hpp"

int main(int argc, char* argv[]) {
	dart::dynamics::BallJoint::Properties properties;
	return 0;
}
```

When compiled with debug symbols, I get the following output from GDB, crashing on the line that declares `properties`:

```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Program received signal SIGABRT, Aborted.
0x00007ffff601f418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
0x00007ffff601f418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff601f418 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff602101a in __GI_abort () at abort.c:89
#2  0x00007ffff665884d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff66566b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff6656701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff6656919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff6656ebc in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff66e83df in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff79f9bfa in dart::dynamics::detail::GenericJointUniqueProperties<dart::math::SO3Space>::GenericJointUniqueProperties(dart::dynamics::detail::GenericJointUniqueProperties<dart::math::SO3Space> const&) () from /usr/lib/libdart.so.6.1
#9  0x00007ffff79ef490 in dart::dynamics::BallJoint::Properties::Properties(dart::dynamics::detail::GenericJointProperties<dart::math::SO3Space> const&) ()
   from /usr/lib/libdart.so.6.1
#10 0x00000000004141ba in main (argc=1, argv=0x7fffffffd948)
```

I am running Ubuntu 16.04.1 LTS (xenial) with the `libdart6` series of packages installed via PPAs as described in the documentation.  The version of `libdart6` is 6.1.0.ppa1~xenial.

Is there any advice in resolving this issue?

jslee02, 2016-10-14:
Thanks for the reporting! It seems this is related to #780, and should be resolved by #790. The patch will be included in the release of DART 6.1.1. In the meantime, you could use DART of branch `release-6.1` that includes the patch.

jslee02, 2016-10-15:
As a quick note, the debian packages of DART 6.1.1, which contains the patch #790, are uploaded to the PPA. Let us know if it works for you!

jslee02, 2016-10-17:
Addressed by #790

closed 2016-10-17 by jslee02