dartsim/dart #798: Failure of testOptimizer with release-5.1 (tag 5.1.4)

tianyang9310, 2016-10-16:
I still cannot pass testOptimizer with release-5.1( tag 5.1.4 )
```
[==========] Running 5 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 5 tests from Optimizer
[ RUN      ] Optimizer.GradientDescent
[       OK ] Optimizer.GradientDescent (0 ms)
[ RUN      ] Optimizer.BasicNlopt
[       OK ] Optimizer.BasicNlopt (2 ms)
[ RUN      ] Optimizer.BasicIpopt
Error [testOptimizer.cpp:185] Ipopt does not pass this test yet. Please see #153.[       OK ] Optimizer.BasicIpopt (0 ms)
[ RUN      ] Optimizer.InverseKinematics
[       OK ] Optimizer.InverseKinematics (2 ms)
[ RUN      ] Optimizer.OutStream
[       OK ] Optimizer.OutStream (1 ms)
[----------] 5 tests from Optimizer (5 ms total)

[----------] Global test environment tear-down
[==========] 5 tests from 1 test case ran. (5 ms total)
[  PASSED  ] 5 tests.
```

jslee02, 2016-10-17:
It actually passed, but just prints an error message that says the ipopt test is disabled. It is misleading, though. By the way, [the error message](https://github.com/dartsim/dart/blob/v5.1.4/unittests/testOptimizer.cpp#L185) should have been removed by #766. I'll create an PR for this.

The actual issue is that [`ipopt` of Ubuntu 14.04](http://packages.ubuntu.com/trusty/coinor-libipopt-dev) (v3.11.4) seems to have [a problem](https://travis-ci.org/dartsim/dart/jobs/168146121#L1790). I could pass the same test with [`ipopt` of Ubuntu 16.04](http://packages.ubuntu.com/xenial/coinor-libipopt-dev) (v3.11.9). That said, you should use `ipopt` on Xenial or use `ipopt` 3.11.9 or higher built from source.

jslee02, 2016-10-17:
Addressed by #799 and #800.

closed 2016-10-17 by jslee02