dartsim/dart #804: Rendering - white lines for non-osg display with nvidia driver

fredowski, 2016-10-24:
Hi, 

i compiled the current DART git version "last commit: Update Control - 19.10.2016" and tried some of the examples. I noticed that there is a difference in the rendering result between the osg examples and the non-osg examples. I think there is a difference in the OpenGL configuration. My system setup is debian jessie NVidia GTX 1080 with native NVidia Driver version 367.48. 

Here is the non-osg atlasSimbicon on NVidia GTX1080 with white lines:

[image]

Here is osgAtlassimbicon on GTX1080  with no white lines:
[image]

Jeongseok figured out that the white lines are not visible with a Intel Integrated Graphics Controller:

[image]

fredowski, 2016-10-24:
Hi Jeongseok,

i think I found the problem regarding the strange white lines. With pullrequest #805 this should be fixed. With the change there are no white lines anymore.

Regards

Friedrich

jslee02, 2016-10-24:
Awesome! Resolved by #805.

closed 2016-10-24 by jslee02