dartsim/dart #810: Consider using #pragma once instead of (#ifndef, #define, #endif)

jslee02, 2016-11-02:
As far as I know, the only reason `#pragma once` is not preferred is that it's not C++ standard. However, [it's supported by most compilers](https://en.wikipedia.org/wiki/Pragma_once#Portability). Using `#pragma once` would allow us to save time by typing less.

mxgrey, 2016-11-02:
The choice between ``#pragma once`` and ``#ifndef`` is an interesting topic, which I don't have any particularly strong personal feelings about.

I think the main arguments against ``#pragma once`` are
1. If there are two copies of a file, then ``#pragma once`` could allow both of them to be included.
2. If you have any kind of unusual file system usage (like remote mounts or file links) using ``#pragma once`` can confuse the compiler.

For case (1), I think if the compiler is looking at two different copies/version of the same file at the same time, then we have bigger problems than what header guard method we're using, so I don't see that as a deal breaker.

For case (2), I really don't know how likely this is to be a problem. For me personally I don't see it mattering, but I wouldn't want to alienate a user who might have an unusual setup. I think if anything might be deal breaker, it would be this.

I do like the idea of not needing gross ``DART_NAMESPACE_FILENAME_H_`` tags on every file. I think it's just a question of whether we need to be worried about argument (2).

jslee02, 2016-11-03:
Hm, I don't think that's usual case that we use symbolic links when building DART (not the consuming project). Is there a case using symbolic links is useful?

mkoval, 2016-11-03:
I am generally wary of using`#pragma once` because it is not standard and the alternative (using include guards) is not too burdensome. The [Google style guide](https://google.github.io/styleguide/cppguide.html) agrees. I don't feel *too* strongly about this, but I do have a preference for keeping the include guards.

mxgrey, 2016-11-03:
Is it plausible that someone might host the source code remotely and then build on a local machine while referring to the remote source during the build process? If so, would that cause issues with ``#pragma once``?

I genuinely do not know the answers to either of those questions, but I think if the answers to both are "yes", then we may need to stick with standard include guards.

jslee02, 2016-11-03:
I used not to prefer `#pragma once` since it's not standard. However, after I realized that it's widely supported by many compilers, I started to reconsider this because it bothers me by keeping updating the header guards whenever the header name is changed or moved to a different directory. By the way, I don't have a strong preference _yet_.

Hm, I also don't know the answers to the remote source building cases.

Okay, as the current way doesn't have problems and doesn't have a strong reason for switching to `#pragma once`, I think we could stick with the header guard.

mkoval, 2016-11-03:
All of these tricky issues may be why `#pragma once` have not been standardized. 😉

closed 2016-11-05 by jslee02