dartsim/dart #815: Consider renaming namespace `utils` to `io`

jslee02, 2016-11-03:
`io` (stands for Input/Output) would be more relevant name according to the code.

mkoval, 2016-11-03:
I generally like this idea. If there are parts of `utils` that are not related to I/O, perhaps we could move them to `common`? I don't see a strong distinction between `utils` and `common` as it stands.

jslee02, 2016-11-03:
> If there are parts of utils that are not related to I/O, perhaps we could move them to common?

I think so. I 

>  I don't see a strong distinction between utils and common as it stands.

They were originally one directory as utils. I seperated out common from utils in the process of arranging the dependencies between the subdirectories (we didn't even have namespaces at that time). After that, it turned out utils only contains IO related functionalities.

psigen, 2016-11-03:
This seems like a reasonable refactor to me.  I think it's a good idea to have an `io` namespace in general.

PR #924 (2017-10-23): Rename utils namespace to io
Depends on #923 
Resolve #815
closed 2017-10-23 by jslee02