dartsim/dart #819: Apply force to an arbitary point of robot body

AbbasKarami, 2016-11-06:
Hi every body,
I really appreciate if anybody can help me with the following issue:
I am simulating a robot using its "urdf" model in "dart". I want to apply an arbitrary amount of force to an arbitrary point in robot body. Is it possible? Does any body have any suggestion about how to accomplish that?
In the tutorial, external force applied to body node using "dart::collision::CollisionNode::getBodyNode". I want to know:
- Is it possible to do the same thing for a "urdf" model (I mean to get a body node and apply force to that)?
- Is it possible to define arbitrary nodes an robot body, when we use a "urdf" model?
Thanks.

jslee02, 2016-11-06:
> Is it possible to do the same thing for a "urdf" model (I mean to get a body node and apply force to that)?

You can find a body from a skeleton if you know the name of bode. For example, if you want to get a body named "left_arm" in the urdf file, you could:
```cpp
auto leftArm = skel->getBodyNode("lef_arm");
```
Then you could apply force or torque to the body:
```cpp
leftArm->setExtForce(
  Eigen::Vector3d(1, 0, 0), // linear force expressed in world coordinates
  Eigen::Vector3d(0.1, 0.1, 0.1)  // offset from the origin of the body frame that you apply the linear force at. The offset is expressed in local coordinates
);
leftArm->setExtTorque(
  Eigen::Vector3d(0, 1, 0)  // torque (or angular force) expressed in world coordinates
);
```

> Is it possible to define arbitrary nodes an robot body, when we use a "urdf" model?

If you refer to "finding" an arbitrary body from a skeleton that was loaded from a "urdf", then you could do that using `Skeleton::getBodyNode("THE_NAME_OF_BODY")`.

Hope this helps.

AbbasKarami, 2016-11-06:
Thank you for your complete answer, as I understand from the first part of your explanation:

leftArm->setExtForce(
  Eigen::Vector3d(1, 0, 0), // linear force expressed in world coordinates
  Eigen::Vector3d(0.1, 0.1, 0.1)  // offset from the origin of the body frame that you apply the linear force at. The offset is expressed in local coordinates
);
leftArm->setExtTorque(
  Eigen::Vector3d(0, 1, 0)  // torque (or angular force) expressed in world coordinates
);

we can apply force to any point of the body (if we have its coordination in the body frame). It is regardless of body nodes is it correct?

jslee02, 2016-11-06:
Yes, it's correct. You also can specify the reference coordinates of the forces and the offset using the additional boolean flags. Please find more information here:
* [setExtForce()](https://dartsim.github.io/api/v6.1.1/d5/d2e/classdart_1_1dynamics_1_1BodyNode.html#aecadf275cda4f6cad395e4e0f5fc58a5)
* [addExtForce()](https://dartsim.github.io/api/v6.1.1/d5/d2e/classdart_1_1dynamics_1_1BodyNode.html#a4332e12156d1845039b89b05eb0c136e)
* [setExtTorque()](https://dartsim.github.io/api/v6.1.1/d5/d2e/classdart_1_1dynamics_1_1BodyNode.html#a1ec55b0a0bf4a06a85c521f7705a50a3)
* [addExtTorque()](https://dartsim.github.io/api/v6.1.1/d5/d2e/classdart_1_1dynamics_1_1BodyNode.html#ad58bad9a2b71b4f216a0c5da6a7f3efc)

nabafa, 2016-12-14:
Hello everybody

I want to know if there is any method to obtain the force in an arbitrary point  on robot body (vice versa of previous discussion). For example in each of the humanoid feet during simulation (like when we have a force sensor there).  Or when manipulator is pushing something in the end effector.
I have found :
dart::dynamics::BodyNode::getExternalForceGlobal ( ) 
Is this correct?

jslee02, 2017-01-19:
I guess you refer to contact force. You could get contact forces using `World::getLastCollisionResult()` after each simulation step.

`getExternalForceGlobal()` returns the sum of all the external forces exerted by the user (e.g., using `BodyNode::set/addaddExtForce`) in the generalized coordinates, which I believe this is not what you're looking for.

nabafa, 2017-01-19:
You are right, I want the contact force in a special point while the robot may have some other contacts. I checked the function you mentioned but I do not find out how to use it for a special point. It seems that it returns the collision force in the defined world. Is it correct?
Should I define a separate world for the robot part in collision to use this function?

jslee02, 2017-01-19:
In DART, contact manifold is approximated by multiple points. For example, the contact manifold of a box on the ground would be the face of the box touching the ground. You may want to get the contact force of an arbitrary point in the face, but it can't be directly computed by DART. Instead, the contact face is approximated by four contact points (typically the points are placed at the vertices of the face, but this could be varied depending on the underlying collision detection engines: fcl, bullet), and you get the contact forces at the contact points.

You may need to compute the actual contact forces that are assumed to be uniformly distributed over the contact face yourself using the contact forces obtained from DART.

nabafa, 2017-07-22:
Hi every body

when I use "World::getLastCollisionResult()" and then I use the function "getContacts", I see 14 contacts in my scene. While "collisionObject1" and "collisionObject2" shows just three objects have contact with each other (in 7 contacts, object A contacts with B and in 7 other contacts object A contacts with C). Clearly, I obtain 14 force vectors by calling "force" for the "getContacts" output vector. I want to compute the contact force between each of these two contacting objects (A with B (F_{AB}) and A with C (F_{AC})) .Is it correct to compute the mean value for  7 contacts regarding object A contacting with B for F_{AB} and the other 7 data for object A contacting with C F_{AC}?

jslee02, 2017-08-31:
If you would like to obtain the net force, you should compute the sum, not the average. The contact forces are calculated to satisfy the unilateral contact constraints assuming all the forces at the contact points that are generated from the collision detector are exerted. Also, the contact point locations should be taken into consideration when you are summing the forces.

jslee02, 2018-02-13:
Closing this issue lack of activity. Please reopen if have further issues.

closed 2018-02-13 by jslee02