dartsim/dart #820: Force actuators

costashatz, 2016-11-11:
I would like to know what exactly the force actuators take as input. Is it the torque? Or something else?

I have a small and light arm (4-DOF -- less than 200 grams each link [including the actuators]) and I put to each actuator (in the URDF) proper maximum torques (a little more than the stall torques of the actual actuators). However, this torque in DART seems insufficient to make it go to a position, which it is surely not likely (i.e. it should be enough to make it move).

Is there a bug when reading the URDF or the force actuators do not take torque as input? I am using standalone DART (without Gazebo).

mxgrey, 2016-11-11:
The torque limit given to the model should include any gear reduction in the joint rather than be the plain motor's torque limit. That could be one explanation.

costashatz, 2016-11-11:
> The torque limit given to the model should include any gear reduction in the joint rather than be the plain motor's torque limit. That could be one explanation.

So they expect torque right? I will check the motor specs in detail if they are expecting torque..

mxgrey, 2016-11-11:
I using "torque" as a generic term for actuator force. All I really mean is that the model needs the **joint** force limit, not the raw motor limit. If there is any other mechanism in the joint that affects the final output force, like gearing, then that mechanism should be accounted for in the joint force limit.

jslee02, 2016-11-11:
One thing you should make sure is that you're applying the joint force every time step. By default, the input force is reset to zero after every step is processed.

costashatz, 2016-11-11:
> One thing you should make sure is that you're applying the joint force every time step. By default, the input force is reset to zero after every step is processed.

Yes I am doing it at every step..

> I using "torque" as a generic term for actuator force. All I really mean is that the model needs the joint force limit, not the raw motor limit. If there is any other mechanism in the joint that [...]

OK. I get it.. Thanks!

closed 2016-11-11 by costashatz