dartsim/dart #821: Degrees of freedom index

nabafa, 2016-11-18:
Hi everybody,
I am using a 27DOF urdf (27 revolute joints)in dart but when I get the number of DOF from dart, it finds 33 degree of freedom. I thought that it is related to the "fixed joints", since there are 6 fixed joint definition in urdf file. However I can not obtain their index in skeleton using "getIndexInSkeleton()". I have obtained the index for "revolute joints" which are between 6-32.
- Does anybody know if the index is skeleton is the same as index which the inertia matrix etc. are constructed according to that?
- Is it possible to find the source of the other DOF definition? and Is it possible to omit them form DOF definition?

Thanks.

mkoval, 2016-11-18:
DART inserts a 6 DOF `FreeJoint` between the URDF and `WorldFrame`. This is necessary to simulate any kind of mobile robot that moves relative to the world frame. If your robot is stationary, then you can use `changeParentJointType` to change the root joint to a `FixedJoint`, which is 0 DOF.

I believe the index of a DOF in the `Skeleton` does match the indexing scheme used for the inertia matrix, Jacobian, etc.

nabafa, 2016-12-11:
Thanks for your reply "mkoval".
Is there any documentation about how the dynamic model for floating base robot is derived in "DART"?
What is the algorithm and dynamic equations?
I want to know if the employed algorithm is the same as the  one discussed for floating base robots in "Rigid body dynamics algorithm" by "Featherstone".
Any body can help me?

mxgrey, 2016-12-11:
Yes, DART uses Featherstone's Articulated Body Algorithm when computing forward and inverse dynamics. That also applies to the dynamics of a floating base robot.

nabafa, 2016-12-11:
Thanks "mxgrey".
Do you know if there is any method to obtain the floating-base joint frame?
Does any function compute composite inertia of the whole robot? (I am not sure if "std::array<double,6> dart::dynamics::Inertia::mMoment" give exactly this parameter.)
Or generally the decoupled parameter stated in "Rigid body dynamics algorithm" by "Featherstone".

mxgrey, 2016-12-13:
> Do you know if there is any method to obtain the floating-base joint frame?

In DART, we define a frame per BodyNode. You can obtain the transform of the floating-base body frame using:

```
Eigen::Isometry3d base_tf = skeleton->getBodyNode(0)->getWorldTransform();
```

or equivalently

```
Eigen::Isometry3d base_tf = skeleton->getRootBodyNode()->getWorldTransform();
```

If you want the joint transform specifically, you can do the following:

```
const BodyNode* bn = skeleton->getBodyNode(0);
const Eigen::Isometry3d T_body = bn->getWorldTransform();
const Eigen::Isometry3d T_child_to_joint = bn->getParentJoint->getTransformFromChildBodyNode();
Eigen::Isometry3d base_joint_tf = T_body * T_child_to_joint;
```

However, in many cases this joint transform will be equivalent to the BodyNode transform, since the transform between the joint frame and its child body is usually just an identity matrix.

> Does any function compute composite inertia of the whole robot?

Correct me if I'm misinterpreting, but it sounds like you are asking about a function that computes the moment of inertia of the whole around its center of mass using something like parallel axis theorem, under the assumption that all joints are locked in their current positions. Off the top of my head, I don't think we have that, but we do offer a helper function for performing the parallel axis theorem. You could use that with a for-loop to do this pretty easily. I believe it would look something like this:

```
Eigen::Matrix3d inertia = Eigen::Matrix3d::Zero();
const Eigen::Vector3d center = skeleton->getCOM();
for(size_t i=0; i < skeleton->getNumBodyNodes(); ++i)
{
  const BodyNode* bn = skeleton->getBodyNode(i);
  const Eigen::Vector3d offset = bn->getCOM() - center; // This is in world coordinates
  const Eigen::Matrix3d R = bn->getWorldTransform().linear();
  const Eigen::Matrix3d I = bn->getInertia()->getMoment(); // This is in body-fixed coordinates
  inertia += dart::math::parallelAxisTheorem(R*I*R.transpose(), offset, bn->getMass());
}
```

> Or generally the decoupled parameters stated in "Rigid body dynamics algorithm" by "Featherstone".

I'm not exactly sure which parameters you mean. You can find the spatial inertial tensor of each rigid body with the following:

```
Eigen::Matrix6d tensor = bodyNode->getInertia()->getSpatialTensor();
```

I know Featherstone also uses a number of intermediary "temporary" parameters during the forward/backward/forward recursion steps, so I'm not sure if you're referring to any of those. If you are, then I think those parameters only exist in a protected cache. If you need access to those for some reason, then I'd be happy to discuss possible options for retrieving them.

nabafa, 2016-12-13:
Thanks for your precise explanation.
If you have access to the "Featherstone" book, I need to construct the relation 9.14 for my robot (floating-base matrices analogues to the coefficients of the standard equation of motion) . I am trying to find the matrices which can be used to construct the mentioned equation. I really appreciate your advises to do this!

jslee02, 2017-01-19:
`Skeleton::getMassMatrix()` returns the first matrix in (9.13), which is the joint space inertia matrix in the notation of the book, where the dimension is n by n (n is the degrees-of-freedom of the system). Also, you can get the bias term (i.e., [ p_0^c \\ C ]) using `Skeleton::getCoriolisAndGravityForces()`. I believe you can compose any matrix in Example 9.2 using the above joint space inertia matrix and the bias term.

jslee02, 2017-01-19:
I believe we are good to close this issue. Please feel free to reopen if necessary.

closed 2017-01-19 by jslee02