dartsim/dart #824: Questionable results

mottosso, 2016-12-04:
Hi,

I'm evaluating Dart for an upcoming project and found this a little disconcerting.

[image]

It's a recording of the `addDeleteSkels.exe` binary from Dart 5.1.4 (Windows 10, VS2015, 32-bit).

Is it normal for the falling purple cube to drop this way? I would have expected it to gain quite a bit more torque upon colliding with the sitting purple cube.

For clarity, the falling purple cube is this:

[image]

And I would have expected it to move like this, but quicker.

[image]

Also note how cubes surrounding it are steady rotating, when I would have expected them to come to a standstill much sooner. They seem to be fighting gravity on their own behalf.

jslee02, 2016-12-04:
The actual reason the example doesn't look physically correct is that the moments of inertia of the boxes are incorrect. All the inertia are identity ([1 0 0; 0 1 0; 0 0 1]), which is too large for those boxes in the example.

I believe #825 fixes this issue by setting the inertia accordingly.

mottosso, 2016-12-04:
Thanks for the quick response @jslee02, and happy to hear it's a matter of configuration!

Out of curiosity, I'm sure this is obvious to most, but as someone new to rigid body dynamics libraries in general (and with more practical experience as a technical animator in the film industry), why would one want to control the inertia of an object in this way?

mxgrey, 2016-12-04:
> why would one want to control the inertia of an object in this way?

The geometry of an object does not guarantee any particular inertia configuration. For example, if one side of a cube is made of more dense material than the other side, you may get an unexpected inertia matrix.

We want DART to be as general as possible, so we leave it to the user to specify the inertia properties of the rigid bodies rather than making too many assumptions. Of course, we also provide convenience functions to assist users in setting up easy cases.

mottosso, 2016-12-04:
> For example, if one side of a cube is made of more dense material than the other side, you may get an unexpected inertia matrix.

Ok, that makes sense. Thanks @mxgrey, looking forward to playing with this!

I'm happy to close this issue.

closed 2016-12-04 by jslee02