dartsim/dart #827: Unconventional definition of ellipsoid size

ljklonepiece, 2016-12-20:
It seems the input to EllipsoidShape is not the radius but it is the 2*radius (diameter) along each dimension, which is a bit confusing to use. (takes me a while to figure out it is the case)

It will be better if convention is followed:)

psigen, 2016-12-20:
Could you clarify where you have seen the convention of using radius for this?

The convention that I have seen typically is to specify an ellipse by the lengths of its principle axes, e.g.: https://en.wikipedia.org/wiki/Ellipsoid

ljklonepiece, 2016-12-20:
from the wiki page, it uses semi-principal axes to define an ellipsoid.
but anw, the principle axes shall not be termed as the radius, right? take a sphere as an example, its length of principle axis is 2 times of its radius.

psigen, 2016-12-20:
Right, the semi-principal axes are the convention I am familiar with.

You are right that when the semi-principle axis lengths are used they should not be termed the `radii` in any of the function parameters or examples.  Thanks for the clarification!

ljklonepiece, 2016-12-21:
No problem! Thanks for creating and maintaining this great physics simulator for the community:)

jslee02, 2016-12-21:
I agree. For the API compatibility, we might want to add `get/setRadii()` first and then change the constructor in the next major release.

jslee02, 2017-01-16:
I believe this is addressed by #829.

closed 2017-01-16 by jslee02