dartsim/dart #830: Target 'utils' is not a component of dart.

jiapei100, 2016-12-31:
ENV:

> Ubuntu 16.04.1
> GCC 5.4.0
> dart : today's dev (git  clone)

During configuration,
```
 CMake Error at cmake/Components.cmake:112 (message):
   Target 'utils' is not a component of dart.
 Call Stack (most recent call first):
   dart/gui/CMakeLists.txt:30 (add_component_dependencies)

```

How to resolve this dependency?

Cheers
Pei

jslee02, 2017-01-19:
I have no problem building the current master branch in the same environment. Could you let me know how exactly did you build DART (e.g., commands)? Also, providing the whole output during the configuration would be helpful.

cyberguy42, 2017-04-21:
I also got this error while compiling. <WORKSPACE> refers to the root directory of a catkin workspace. Here is my configuration output:

```
-- ============================================
--                 DART 6.2.0
-- ============================================
-- Found FCL: <WORKSPACE>/devel/fcl/include (found suitable version "0.6.0", minimum required is "0.2.9") 
-- Found ASSIMP: /usr/include (found suitable version "3.0.1264", minimum required is "3.0.0") 
-- Performing Test ASSIMP_AISCENE_CTOR_DTOR_DEFINED
-- Performing Test ASSIMP_AISCENE_CTOR_DTOR_DEFINED - Failed
-- Performing Test ASSIMP_AIMATERIAL_CTOR_DTOR_DEFINED
-- Performing Test ASSIMP_AIMATERIAL_CTOR_DTOR_DEFINED - Failed
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Looking for nlopt - NOT found, to use dart-optimizer-nlopt, please install nlopt (>= 2.4.1)
-- Looking for ipopt - NOT found, to use dart-optimizer-ipopt, please install ipopt (>= 3.11.9)
-- Performing Test BT_USE_DOUBLE_PRECISION
-- Performing Test BT_USE_DOUBLE_PRECISION - Success
-- Looking for ode - NOT found, to use dart-collision-ode, please install ode (>= 0.11)
-- Looking for tinyxml2 - NOT found, to use dart-utils, please install tinyxml2 (>= 1.0.1)
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
CMake Warning at cmake/DARTFindDependencies.cmake:282 (message):
  Could NOT find OpenSceneGraph nor OpenThreads -- we will skip dart-gui-osg

  If you believe you do have both OSG and OpenThreads installed, try setting
  OSG_DIR
Call Stack (most recent call first):
  CMakeLists.txt:138 (include)

CMake Error at cmake/Components.cmake:112 (message):
  Target 'utils' is not a component of dart.
Call Stack (most recent call first):
  dart/gui/CMakeLists.txt:30 (add_component_dependencies)

-- Configuring incomplete, errors occurred!

```

jslee02, 2017-04-22:
Thank you for the reporting. I could reproduce the issue based on your configuration output, and am working on it.

cyberguy42, 2017-04-22:
If it helps, I was able to avoid the error by installing tinyxml2.

jslee02, 2017-04-22:
Yes, it helps. The main reason of the error was that DART attempts to build `dart-gui` component while `dart-utils` is not built that is a dependency of `dart-gui`. I believe #880 fixes this.

PR #880 (2017-04-25): Rework on dependency checking
Currently, all the optional dependencies were searched in a batch whereas those dependencies are required by different optional components. I think it would make more sense to move the dependency checks to the each component that requires them.

This also should resolve #830.
closed 2017-04-25 by jslee02