dartsim/dart #837: Humanoid Jacobian  with respect to the support foot

nabafa, 2017-01-24:
Hello everybody

I am looking for a method to obtain the Jacobian matrix of my humanoid robot between two arbitrary points using dart. Using "getLinearJacobian" and other similar function, dart gives us the Jacobian matrix with respect to the virtual floating base joint. But I need the Jacobian of COM or hands or the other foot with respect to the support foot. Is it possible through dart?

I really appreciate any suggestions.
Thanks.

jslee02, 2017-01-26:
DART does not currently provide the exact API for relative Jacobian, but you might be able to obtain it through simple matrix computation. Suppose you want to get the Jacobian of a body1 relative to body2, then the relative Jacobian between them is simply J1 - J2 where J1 and J2 are the Jacobian of body1 and body2, respectively, w.r.t. the virtual floating base joint expressed in the world coordinates. If the Jacobians are expressed in the body frame coordinates, you might need to transform the Jacobians accordingly.

jslee02, 2017-11-12:
Closing due to lack of a response. Please re-open if you have further trouble.

closed 2017-11-12 by jslee02