dartsim/dart #838: Materials from URDF is OSG?

costashatz, 2017-01-25:
Hello,

I have a urdf model of a simple manipulator (with stl meshes). When I am loading in DART and display it with osg, I get the following:

[image]

Note that I am NOT using DART through Gazebo, but as a standalone. Changing [this](https://github.com/dartsim/dart/blob/master/dart/dynamics/MeshShape.cpp#L135) line to ``mColorMode(COLOR_INDEX)``, I get the following:

[image]

Note that the STLs do not contain materials and I am setting the color in the urdf. Although, I did try with materials in the STLs and didn't change anything.

What I should have gotten is something like this:

[image]

Any clues why I get this behavior?

You can find the xacro (which you can use to generate the urdf) [here](https://github.com/resibots/omni_ros/blob/omnigrasper/omni_description/robots/omnigrasper_arm_only.urdf.xacro). The repo also has the STL files [here](https://github.com/resibots/omni_ros/tree/omnigrasper/omni_description/stl)...

Thanks!

costashatz, 2017-03-01:
Any ideas on this?

mxgrey, 2017-03-01:
One thing that jumps out at me while looking at the [URDF](https://github.com/resibots/omni_ros/blob/omnigrasper/omni_description/robots/omnipointer_arm_only.urdf) is that it specifies the material properties in the URDF (which makes sense, since STL doesn't usually include material properties), but our [URDF parser](https://github.com/dartsim/dart/blob/ca61f40764d107c54498d57cee0a8f2d7d915a94/dart/utils/urdf/DartLoader.cpp#L462) appears to put the URDF's material information into the primitive "color" field (which the ``MeshShape`` refers to as ``SHAPE_COLOR``) rather than making it part of the material properties. This could easily lead to some confusion, because in order to see the "material" of a mesh whose material was specified through a URDF, you need to set use ``MeshShape::setColorMode(SHAPE_COLOR)`` (or change the code in the constructor, similar to what you did).

Have you tried using the ``SHAPE_COLOR`` mode rather than ``MATERIAL_COLOR`` or ``COLOR_INDEX``? I'll check this on my end right now, after I retrieve all the files, and if that doesn't help, then I'll need to investigate further.

mxgrey, 2017-03-01:
I'm a bit out of practice with ROS (I've just now installed ROS Indigo to test this issue after not having ROS for ~3 years), so I may need some assistance in order to test it.

I created a catkin workspace, cloned the repo you linked to, checked out the omnigrasper branch, and installed the youbot_driver_ros_interface dependency. However, I still get the following compilation error when I run ``catkin_make``:

```
/home/grey/catkin_ws/src/omni_ros/omni_driver/src/omni.cpp:5:54: fatal error: youbot_driver_ros_interface/BaseDisplace.h: No such file or directory
 #include <youbot_driver_ros_interface/BaseDisplace.h>
```

Is there some more setup that I'm neglecting before the build process can be successful?

Alternatively, is there a way to generate the URDF without needing to run ``catkin_make``? I tried to run 

```
rosrun xacro xacro.py omnigrasper_arm_only.urdf.xacro > omnigrasper_arm_only.urdf
```

from ``~/catkin_ws/src/omni_ros/omni_description/robots``, but I get a Python error which ends with:

```
rospkg.common.ResourceNotFound: omni_description
```

I'm sure there's probably something wrong with my procedure, but it's been so long since I last used ROS that I can't think of what to do differently.

costashatz, 2017-03-01:
It's not your fault. We have some dependencies that are not mandatory. Here's the actual URDF:

[omnigrasper.urdf.zip](https://github.com/dartsim/dart/files/812118/omnigrasper.urdf.zip)

Make sure that you have the following directory structure:

- arm_meshes
  - STL files
- URDF
  - omnigrasper.urdf

If I remember well, I did also `MeshShape::setColorMode(SHAPE_COLOR)` and it had a similar result with what I showed you in the first post.

I will try to test something also.. Thank you for your help..

mxgrey, 2017-03-05:
I've identified the error, however I'm not sure whether it would best be characterized as a bug in urdfdom or a misunderstanding (on the part of DART) about the expected behavior of urdfdom (I think it should be characterized as a bug in urdfdom, because the behavior seems rather illogical to me).

It seems that when urdfdom parses an array of visuals, [it assigns the first entry in the array to a special field in the ``urdf::Link`` which is simply called ``visual``](https://github.com/ros/urdfdom/blob/8b27e0b6fbddbb4cc670156961bd52b4d5ea632b/urdf_parser/src/link.cpp#L503). Then after all the xml information for the ``urdf::Link`` has been parsed, it does a little bit of post-processing where the material of the field ``urdf::Link::visual`` [is checked against the list of materials for the model](https://github.com/ros/urdfdom/blob/8b27e0b6fbddbb4cc670156961bd52b4d5ea632b/urdf_parser/src/model.cpp#L142). However, all of the remaining entries in ``urdf::Link::visual_array`` are **ignored**, and therefore their material fields point to material instances which were initialized with only a name. In my opinion, the behavior of urdfdom would be far more rational if they replaced the line

```
if(link->visual)
```

with something like

```
for(auto& visual : visual_array)
```

And then replace instances of ``link->visual`` with ``visual``. This would give consistent behavior for all entries of ``visual_array`` instead of arbitrarily giving special treatment to the first entry.

I can only assume that other renderers which produce the correct results are either (1) not using urdfdom to parse the urdf file, or (2) do some post-processing where they take it upon themselves to link the names of materials in the visuals to their appropriate entries in the model.

I'll write up a patch which does (2) for DART, but I might also submit a pull request to urdfdom to change the behavior of their parser.

costashatz, 2017-03-05:
Great! Thanks! I am waiting for your patch...!

mxgrey, 2017-03-05:
The fixes are now available in #851. It also includes a fix for issue #676 which you raised almost a year ago. I realized a solution for that particular problem in a moment of inspiration. I apologize for not fixing it sooner.

However, it should be noted that it's still necessary to apply ``MeshShape::setColorMode(dart::dynamics::MeshShape::SHAPE_COLOR);`` to all ``MeshShape`` instances. This can be done with something like the following code:

```
for(size_t i=0; i < skeleton->getNumBodyNodes(); ++i)
{
  dart::dynamics::BodyNode* bn = skeleton->getBodyNode(i);
  for(size_t j=0; j < bn->getNumShapeNodes(); ++j)
  {
    dart::dynamics::ShapeNode* sn = bn->getShapeNode(j);
    if(sn->getVisualAspect())
    {
      dart::dynamics::MeshShape* ms = 
          dynamic_cast<dart::dynamics::MeshShape*>(sn->getShape().get());
      if(ms)
        ms->setColorMode(dart::dynamics::MeshShape::SHAPE_COLOR);
    }
  }
}
```

Of course that much code for something so basic is rather cumbersome. In the future, it would be good to offer a panel in the OSG interface that allows the user to alter all the various rendering properties during run time, but I don't know that anyone has time to implement something like that in the near future.

costashatz, 2017-03-06:
> The fixes are now available in #851. It also includes a fix for issue #676 which you raised almost a year ago. I realized a solution for that particular problem in a moment of inspiration while working on the current issue. I apologize for not fixing it sooner.

Great work! Do not apologize. I am sure you have your plate full of more important tasks!

> Of course that much code for something so basic is rather cumbersome. In the future, it would be good to offer a panel in the OSG interface that allows the user to alter all the various rendering [...]

I think this (writing this "much" code) would OK for most of my needs, so thank you again!

Thanks!

mxgrey, 2017-03-06:
The fix is merged. Please feel free to reopen if you still encounter issues.

closed 2017-03-06 by mxgrey