dartsim/dart #839: Composite import missing in Homebrew

tessmer, 2017-01-25:
Hi,

I just tried to build with Dart.
It wasn't possible because dart/common/detail/AspectWithVersion.hpp was missing the import of Composite: "#include /dart/common/Composite.hpp"

This problem arose with homebrew on macOS Sierra.

jslee02, 2017-01-25:
I could successfully install `dartsim6` (v6.1.2) using homebrew on macOS Sierra myself now.

Could you paste or show the whole error messages you got?

r-t-r, 2017-01-26:
This is the error:

`In file included from bipedtest.cpp:32:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:3:
In file included from /usr/local/include/dart/common/Aspect.hpp:37:
In file included from /usr/local/include/dart/common/Cloneable.hpp:324:
In file included from /usr/local/include/dart/common/detail/Cloneable.hpp:1:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:4:
In file included from /usr/local/include/dart/common/AspectWithVersion.hpp:35:
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:15:30: error: unknown type name 'Composite'
          class CompositeT = Composite,
                             ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:16:45: error: use of undeclared identifier 'NoOp'
          void (*updateState)(DerivedT*) = &NoOp<DerivedT*> >
                                            ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:16:50: error: 'DerivedT' does not refer to a value
          void (*updateState)(DerivedT*) = &NoOp<DerivedT*> >
                                                 ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:14:30: note: declared here
template <class BaseT, class DerivedT, typename StateDataT,
                             ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:16:59: error: expected expression
          void (*updateState)(DerivedT*) = &NoOp<DerivedT*> >
                                                          ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:16:61: error: expected unqualified-id
          void (*updateState)(DerivedT*) = &NoOp<DerivedT*> >
                                                            ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:71:30: error: unknown type name 'Composite'
          class CompositeT = Composite,
                             ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:72:50: error: use of undeclared identifier 'NoOp'
          void (*updateProperties)(DerivedT*) = &NoOp<DerivedT*> >
                                                 ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:72:55: error: 'DerivedT' does not refer to a value
          void (*updateProperties)(DerivedT*) = &NoOp<DerivedT*> >
                                                      ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:70:30: note: declared here
template <class BaseT, class DerivedT, typename PropertiesDataT,
                             ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:72:64: error: expected expression
          void (*updateProperties)(DerivedT*) = &NoOp<DerivedT*> >
                                                               ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:72:66: error: expected unqualified-id
          void (*updateProperties)(DerivedT*) = &NoOp<DerivedT*> >
                                                                 ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:141:1: error: unknown type name 'constexpr'
constexpr void (*AspectWithState<
^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:141:11: error: expected unqualified-id
constexpr void (*AspectWithState<
          ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:148:1: error: unknown type name 'AspectWithState'
AspectWithState<BaseT, DerivedT, StateDataT, CompositeT, updateState>::
^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:148:16: error: expected unqualified-id
AspectWithState<BaseT, DerivedT, StateDataT, CompositeT, updateState>::
               ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:220:1: error: unknown type name 'AspectWithVersionedProperties'
AspectWithVersionedProperties<BaseT, DerivedT, PropertiesDataT,
^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:220:30: error: expected unqualified-id
AspectWithVersionedProperties<BaseT, DerivedT, PropertiesDataT,
                             ^
In file included from bipedtest.cpp:32:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:3:
In file included from /usr/local/include/dart/common/Aspect.hpp:37:
In file included from /usr/local/include/dart/common/Cloneable.hpp:324:
[... 11 lines cut ...]
                                                         ^
/usr/local/include/dart/common/AspectWithVersion.hpp:41:17: note: declared here
template <class DerivedT,
                ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
`

When I fix the missing include I get this error:

`In file included from bipedtest.cpp:32:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:3:
In file included from /usr/local/include/dart/common/Aspect.hpp:37:
In file included from /usr/local/include/dart/common/Cloneable.hpp:324:
In file included from /usr/local/include/dart/common/detail/Cloneable.hpp:1:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:4:
In file included from /usr/local/include/dart/common/AspectWithVersion.hpp:35:
In file included from /usr/local/include/dart/common/detail/AspectWithVersion.hpp:6:
In file included from /usr/local/include/dart/common/Composite.hpp:35:
/usr/local/include/dart/common/detail/CompositeData.hpp:35:10: fatal error: 'Eigen/Core' file not found
#include <Eigen/Core>
         ^
1 error generated.`

r-t-r, 2017-01-27:
I tried uninstalling Dart6, upgrading all homebrew packages (because of the qt4-webkit problem which forced me to uninstall gazebo as well to be able to upgrade brew again) and then to install it again.

And now I got these error trying to compile the final biped tutorial code. Why is not even the simple tutorial code working out of the box?

` clang tutorialBiped.cpp
In file included from tutorialBiped.cpp:32:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:3:
In file included from /usr/local/include/dart/common/Aspect.hpp:37:
In file included from /usr/local/include/dart/common/Cloneable.hpp:324:
/usr/local/include/dart/common/detail/Cloneable.hpp:128:55: error: a space is required between consecutive right angle brackets (use '> >')
  return common::make_unique<MakeCloneable<Base, Mixin>>(*this);
                                                      ^
/usr/local/include/dart/common/detail/Cloneable.hpp:212:1: error: 'auto' not allowed in function return type
auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(
^~~~
/usr/local/include/dart/common/detail/Cloneable.hpp:213:22: error: expected ';' at end of declaration
    const Data& data) -> ProxyCloneable&
                     ^
/usr/local/include/dart/common/detail/Cloneable.hpp:213:23: error: cannot use arrow operator on a type
    const Data& data) -> ProxyCloneable&
                      ^
/usr/local/include/dart/common/detail/Cloneable.hpp:223:1: error: 'auto' not allowed in function return type
auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(
^~~~
/usr/local/include/dart/common/detail/Cloneable.hpp:224:17: error: expected ';' at end of declaration
    Data&& data) -> ProxyCloneable&
                ^
/usr/local/include/dart/common/detail/Cloneable.hpp:224:18: error: cannot use arrow operator on a type
    Data&& data) -> ProxyCloneable&
                 ^
/usr/local/include/dart/common/detail/Cloneable.hpp:234:1: error: 'auto' not allowed in function return type
auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(
^~~~
/usr/local/include/dart/common/detail/Cloneable.hpp:235:33: error: expected ';' at end of declaration
    const ProxyCloneable& other) -> ProxyCloneable&
                                ^
/usr/local/include/dart/common/detail/Cloneable.hpp:235:34: error: cannot use arrow operator on a type
    const ProxyCloneable& other) -> ProxyCloneable&
                                 ^
/usr/local/include/dart/common/detail/Cloneable.hpp:245:1: error: 'auto' not allowed in function return type
auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::operator =(
^~~~
/usr/local/include/dart/common/detail/Cloneable.hpp:246:28: error: expected ';' at end of declaration
    ProxyCloneable&& other) -> ProxyCloneable&
                           ^
/usr/local/include/dart/common/detail/Cloneable.hpp:246:29: error: cannot use arrow operator on a type
    ProxyCloneable&& other) -> ProxyCloneable&
                            ^
/usr/local/include/dart/common/detail/Cloneable.hpp:308:1: error: 'auto' not allowed in function return type
auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::get() const -> Data
^~~~
/usr/local/include/dart/common/detail/Cloneable.hpp:308:72: error: expected ';' at end of declaration
auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::get() const -> Data
                                                                       ^
/usr/local/include/dart/common/detail/Cloneable.hpp:308:73: error: cannot use arrow operator on a type
auto ProxyCloneable<Base, OwnerT, DataT, setData, getData>::get() const -> Data
                                                                        ^
In file included from tutorialBiped.cpp:32:
In file included from /usr/local/include/dart/dart.hpp:33:
In file included from /usr/local/include/dart/common/common.hpp:4:
In file included from /usr/local/include/dart/common/AspectWithVersion.hpp:35:
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:57:3: error: unknown type name 'constexpr'
  constexpr static void (*UpdateState)(Derived*) = updateState;
  ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:57:13: error: expected member name or ';' after declaration specifiers
  constexpr static void (*UpdateState)(Derived*) = updateState;
  ~~~~~~~~~ ^
/usr/local/include/dart/common/detail/AspectWithVersion.hpp:113:3: error: unknown type name 'constexpr'
  constexpr static void (*UpdateProperties)(Derived*) = updateProperties;
  ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.`

jslee02, 2017-02-01:
It seems you're building tutorials independently from DART source tree. We've only considered that the tutorials are built in DART source tree.

I've created a [pull request](#842) to enable building tutorials and examples out of DART source tree hoping it addresses this issue.

jslee02, 2017-11-12:
Closing due to lack of a response. Please re-open if you have further trouble.

closed 2017-11-12 by jslee02