dartsim/dart #848: Need more explanation on setting friction coefficient

ljklonepiece, 2017-02-27:
for function 
`
void dart::dynamics::BodyNode::setFrictionCoeff	(	double 	_coeff	)	
`
it sets friction coefficient on one object, but normally friction coefficient is defined between two surfaces.
Shall I interpret this function to set the friction coefficient between this object and any objects? 
but for example A and B are in contact,
`body_A->setFrictionCoeff(0.5)`
`body_B->setFrictionCoeff(1.0)`

What should be the resulting friction coeff between A and B?

jslee02, 2017-02-27:
[DART uses the minimum value when they are different](https://github.com/dartsim/dart/blob/master/dart/constraint/ContactConstraint.cpp#L97). Also, I agree that this should be documented (at least) on `setFrictionCoeff()`, and will do this soon.

mxgrey, 2017-02-27:
It makes sense that some users may want to define friction properties pairwise, since that is more technically accurate.

I wonder if we could implement a ``FrictionFilter`` similar to the ``CollisionFilter`` which might look something like

```
using FrictionFilter = std::function<double(const BodyNode* bn1, const BodyNode* bn2)>;
```

Then if ``FrictionFilter`` is not set, we fall back on the current method.

jslee02, 2017-02-27:
I like the idea! I would prefer `FrictionFunction` though.

mxgrey, 2017-02-27:
That's definitely a better name. Using "filter" was mostly to relate it to the existing ``CollisionFilter``.

jslee02, 2017-02-27:
Sure, I presumed so. :smile:

jslee02, 2017-11-12:
Closing due to lack of a response. Please re-open if you have further trouble.

closed 2017-11-12 by jslee02