dartsim/dart #850: DART is incompatible with FCL 0.6 (current master of FCL)

mxgrey, 2017-03-02:
[DART](https://github.com/dartsim/dart/blob/ca61f40764d107c54498d57cee0a8f2d7d915a94/dart/math/MathTypes.hpp#L65) and [FCL 0.6](https://github.com/flexible-collision-library/fcl/blob/2e3d262195fbfd5369e5aa00fb6c6c7c0ae855de/include/fcl/common/types.h#L124) both contain identical definitions of ``Eigen::aligned_vector`` and a few other similar data types. This causes a definition collision during compilation.

While the motivation behind these names is sensible, it highlights the dangers of defining things within external namespaces. My suggestion would be that DART and FCL rename these types as ``dart::aligned_vector`` and ``fcl::aligned_vector`` or something similar. The two will be interoperable, but giving them separate names will allow compilation.

This is low priority since there is no strict need to use FCL 0.6 with DART, but we'll want to fix this eventually.

a-price, 2017-04-11:
FCL 0.6 has had a pretty significant refactor, so there's a lot more than just the aligned_vector code that needs to be changed. The new FCL code is highly templated, and the name of e.g. the main vector class has changed. I've created a [branch](https://github.com/a-price/dart/tree/fcl-0.6) over in my fork with the necessary changes to get things to work with 0.6, but it's not backwards-compatible with 0.5. Probably what would need to happen would be to have the [FCLTypes](https://github.com/dartsim/dart/blob/master/dart/collision/fcl/FCLTypes.hpp) file typedef a bunch of fcl types according to the version number.

jslee02, 2017-11-12:
I believe this is resolved by #873 and #936.

closed 2017-11-12 by jslee02