dartsim/dart #857: dart crashed when delete an object from gazebo

QingkaiLu, 2017-03-20:
Hi,

I am using dart in gazebo. I need to add and delete objects in gazebo. It works fine to add the object. But whenever I delete the object, dart will crash with the error message: "***** Internal Program Error - assertion (dartLink2.get() != __null) failed in virtual void gazebo::physics::DARTPhysics::UpdateCollision():
/tmp/gazebo/gazebo/physics/dart/DARTPhysics.cc(157): dartLink2 in collision pare is NULL".

I used the gazebo SpawnModel to add and delete objects (http://gazebosim.org/tutorials?tut=ros_comm#SpawnModel). For example, add object: `rosrun gazebo_ros spawn_model -database coke_can -sdf -model coke_can1 -y 2.2 -x -0.3`, delete object: `rosservice call gazebo/delete_model '{model_name: cokecan1}'`

Thank you in advance for any help!

mxgrey, 2017-03-20:
For the record, this appears to be a bug in Gazebo, not DART.

My impression is that there is some collision data in Gazebo which may linger after an object has been deleted. Gazebo is doing a sanity check to ensure that the relevant objects still exist before using that collision data (which is a sensible thing to do).

I don't know much about the internals of Gazebo, but I can think of a few explanations for why you are encountering this. People who have more experience with (or knowledge of) Gazebo are encouraged to correct me:

1. Are you deleting objects while a simulation is actively running? If so, you could easily encounter a race condition where Gazebo is computing the physics of an object's collision, but you are deleting the object out from under the simulator before it's finished computing. If this is the case, you might be able to safely delete the object if you pause the simulation first. Also, it would probably be good for Gazebo to update their documentation to include a warning about this possibility.

2. Perhaps Gazebo is not properly clearing the relevant contact data of an object once it has been removed. If this is the case, there is not much that you, as a user, can do. Someone would have to patch this behavior in the source code of Gazebo.

jslee02, 2017-03-20:
Could you [create an issue on the Gazebo repo](https://bitbucket.org/osrf/gazebo/issues/new) as well?

QingkaiLu, 2017-03-20:
I tried to delete the object in gazebo without using dart (uses the default ODE physics engine of Gazebo instead), which works well. So I assume it's not because of gazebo.

jslee02, 2017-03-21:
Gazebo uses DART but through [the glue code](https://bitbucket.org/osrf/gazebo/src/0d39ebf2b92ba1833f83ea14df668b50f89b5017/gazebo/physics/dart/?at=default). It could be the issue of the glue code or DART itself, or even it's a known issue. Either way, they (and I) still might be interested in this issue.

Also, it would be very nice if you let us know which version or branch are you using because there were some major changes in the glue code to support different version of DART.

QingkaiLu, 2017-03-21:
I am using Gazebo 7 with Dart 4. This is how I installed Dart 'sudo apt-get install libdart-core4-dev'. Let me know if you need more information. 

Thank you for looking into this issue.

QingkaiLu, 2017-03-21:
Since it can be the Gazebo glue code, I also created [an issue in Gazebo Repo](https://bitbucket.org/osrf/gazebo/issues/2234/dart-physics-engine-crashed-after-deleting).

scpeters, 2017-03-21:
Yes, this looks like an issue with gazebo's integration with Dart. Let's continue the discussion on the gazebo issue tracker:

* https://bitbucket.org/osrf/gazebo/issues/1723/dart-link-destruction-segfault

jslee02, 2017-11-12:
Closing in favor of the issue on the Gazebo issue tracker: https://bitbucket.org/osrf/gazebo/issues/1723/dart-link-destruction-segfault

closed 2017-11-12 by jslee02