dartsim/dart #860: Incorrect contact points

a-price, 2017-03-23:
I'm getting some pretty inaccurate contact points when simulating mesh-mesh collisions (shown below). Is this likely a byproduct of flexible-collision-library/fcl#106? If so, is there an option to enable the Bullet collision detector mentioned in #156?

(Yellow lines in the images represent the contact normals. Purple lines represent forces. The right-hand windows are a top-down view of the scenes on the left, with the lego brick replaced by its support outline.)

[image]

[image]

jslee02, 2017-03-23:
> Is this likely a byproduct of flexible-collision-library/fcl#106?

I believe so if the triangles of the mesh is sufficiently large (larger than the half of the longer side of the block).

You could switch to the bullet collision detector by passing it to the constraint solver within the world. Here is an [example code](https://github.com/dartsim/dart/blob/ad7544370669c6088337ef3bcae3a694cf6d7e99/tutorials/tutorialBiped-Finished/tutorialBiped-Finished.cpp#L474-L475).

a-price, 2017-03-23:
Thanks for the suggestion. Unfortunately, when I switch to the Bullet collision detector, no collisions are detected at all. This could be because I had to switch from dart v5.x to v6.x, and may not have transferred everything correctly, but I was seeing the same behavior in 5. Any suggestions here?

I'll keep experimenting with it and report back.

mxgrey, 2017-03-23:
I remember this issue existing: #717

Are you relying on group-group collision checks? Those don't currently have proper support when using bullet.

a-price, 2017-03-23:
I'm basing it off the example in https://github.com/dartsim/dart/blob/v6.1.0/tutorials/tutorialDominoes-Finished.cpp

    dart::collision::CollisionDetectorPtr collisionEngine = mScene->mpWorld->getConstraintSolver()->getCollisionDetector();
    dart::collision::CollisionGroupPtr collisionGroup = mScene->mpWorld->getConstraintSolver()->getCollisionGroup();
    std::unique_ptr<dart::collision::CollisionGroup> objectGroup = collisionEngine->createCollisionGroup(mScene->mpTarget.get());
    // Also tried colliding the hand with the object
    // std::unique_ptr<dart::collision::CollisionGroup> handGroup = collisionEngine->createCollisionGroup(mScene->mpManipulator.get());

    dart::collision::CollisionOption option;
    dart::collision::CollisionResult result;

    bool collision = collisionGroup->collide(objectGroup.get(), option, &result);

As I mentioned, I'm just transitioning away from v5, so I'm still getting the hang of the new collision interface.

a-price, 2017-03-23:
If it's relevant, the object in question is being created like this:

    dart::dynamics::SkeletonPtr target = dart::dynamics::Skeleton::create("target");
    // Give the target a body
    dart::dynamics::BodyNode* body = target->createJointAndBodyNodePair<dart::dynamics::PlanarJoint>(nullptr).second;
    // Give the body a shape
    dart::dynamics::MeshShapePtr shape = std::make_shared<dart::dynamics::MeshShape>(Eigen::Vector3d::Ones(),
                                                                                     dart::dynamics::MeshShape::loadMesh(parsePackageURL(modelPath)),
                                                                                     modelPath);

    dart::dynamics::ShapeNode* sn = body->createShapeNodeWith<dart::dynamics::VisualAspect, dart::dynamics::CollisionAspect, dart::dynamics::DynamicsAspect>(shape);

mxgrey, 2017-03-23:
If you're basing it off of the dominoes tutorial, then it's a single-group collision check, so issue #717 would not explain the issue you're having. The way you're creating the object should be perfectly fine.

If bullet continues to give you issues and the problem with FCL is really related to triangle size, then it may be possible to densify the mesh using Blender or 3DS Max (which is available to students for free). It's not a very elegant solution to this (especially since this shouldn't be an issue in the first place), but if you just need something quick and dirty, it might work.

a-price, 2017-03-27:
I think the issue might be related to using non convex shapes in Bullet, e.g. btGImpactMeshShape. I'm going to play around with using btConvexHullShape and see if that behaves nicer.

a-price, 2017-04-14:
For what it's worth, I implemented a completely different triangle-triangle intersection algorithm in FCL, based on [Möller 1997](https://cis.temple.edu/~lakaemper/courses/cis350_2004/etc/moeller_triangle.pdf). It's not perfect yet, but it does seem to give much more accurate intersections without slowing down too much. I haven't benchmarked it or anything yet, but it does pass the FCL unit tests. The FCL fork is [here](https://github.com/a-price/fcl/tree/custom_triangle_intersect), and currently requires the DART fork [here](https://github.com/a-price/dart/tree/fcl-0.6) to work.

jslee02, 2017-04-14:
Thanks for working on this issue for both side of FCL and DART. I'll take a look those PRs soon.

a-price, 2017-04-14:
No problem. I need pretty accurate contact simulations if I'm ever going to finish my thesis, anyway.

Also, relevant screenshot.

[image]

jslee02, 2025-11-24:
Resolved by https://github.com/dartsim/dart/pull/2258