dartsim/dart #869: DART will wrongly prompt warning for setting initial joint positions out of range

jyf588, 2017-04-04:
Hello everyone,

I had this joint in my Skel file:

>             <joint type="prismatic" name="j_gripper2">
>                 <parent>forearm</parent>
>                 <child>gripper2</child>
>                 <axis>
>                     <xyz>0.0 0.0 -1.0</xyz>
>                     <limit>
>                         <lower>0.01</lower>
>                         <upper>0.10</upper>
>                     </limit> 
>                 </axis>
>                 <init_pos>0.03</init_pos>
>             </joint>

When I ran the simulation, DART gave me the following warning: 

>[1;33mWarning [GenericJoint.hpp:1289][0m [GenericJoint::setRestPosition] Value of _q0 [0], is out of the limit range [0.01, 0.1] for index [0] of Joint [Joint].

Actually I can see from my window that the init_pos is indeed correctly set to 0.03 instead of 0.0. I don't know why in function GenericJoint::setRestPosition(), it will think my init_pos is 0.0

Thanks,
Yifeng

mxgrey, 2017-04-04:
The rest position refers to the "zero spring potential" position that gets used when computing spring stiffness. The default value for this is zero.

Admittedly, this warning might be overly cautious, since many users probably don't even use the spring stiffness. Also, it's plausible that there are situations where the rest position is desired to be outside the position limits. We could consider removing this warning.

jyf588, 2017-04-04:
I see... The warning refers to the unspecified <spring_rest_position>, which has default 0 value. It has nothing to do with my <init_pos> value.

mxgrey, 2017-04-04:
Exactly. I'm glad you brought this up, though, because I think it's worthwhile to consider removing this warning.

PR #1418 (2019-09-20): Allow to set joint rest position out of joint limits
This PR is a rework of #1374.

Resolves #869 

***

**Before creating a pull request**

- [ ] Document new methods and classes
- [x] Format new code files using `clang-format`

**Before merging a pull request**

- [x] Set version target by selecting a milestone on the right side
- [ ] Summarize this change in `CHANGELOG.md`
- [x] Add unit test(s) for this change
closed 2019-09-20 by jslee02