dartsim/dart #888: Unable to build DART tutorial out of source tree on MacOS

jyf588, 2017-05-20:
Hello everyone,

I was trying to build a tutorial out of DART source tree following instructions in dart/tutorials/README.md. But I got this error message from Cmake log: Could NOT find Dart (missing:  DART_ROOT).

I am using OSX, and I'm not sure whether this is related to collision with another "Dart testing software": https://github.com/Kitware/CMake/blob/master/Modules/FindDart.cmake 

How should I resolve this?

Thanks,
Yifeng

jslee02, 2017-05-20:
Did you install DART before building the tutorial?

jyf588, 2017-05-20:
Yes, I think so... I tried both installing from homebrew and installing from source.

jslee02, 2017-05-20:
I could [reproduce the build error](https://travis-ci.org/dartsim/dart-examples/jobs/234309399#L1604). Unfortunately, I don't have macOS for now. Let me take a look at this coming Monday.

jslee02, 2017-05-21:
I believe this should be addressed by #889.

jyf588, 2017-05-23:
My Cmake can find DART now, but I am getting another error during make: 

In file included from /usr/local/include/dart/config.hpp:5:
/usr/local/include/eigen3/Eigen/Core:273:10: fatal error: 'type_traits' file not found
#include <type_traits>

Any help with this? Cmake outputs "Found DART: /usr/local/include (Required is at least version "6.2.0") found components:  utils-urdf gui dart"; is the components complete?

jslee02, 2017-05-23:
That's odd. [`type_traits` is a standard library header](http://en.cppreference.com/w/cpp/header/type_traits). Can you provide more detail on how could I reproduce it (such as how you installed DART, which version or branch or tag, which tutorial, etc)?

jyf588, 2017-05-23:
I installed DART using "sudo make install" from latest DART source (Master branch). (I might previously also installed earlier versions of DART through homebrew; I don't see these two conflict though.)

I was making the BipedFinished tutorial; but the error from config.hpp actually referred from dart.hpp, from my make error log.

Thanks.

jyf588, 2017-05-23:
Add "add_compile_options(-stdlib=libc++)" in Cmake will clear the type_traits error but give me other errors like "libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9"...

Is this the right way to force compiler to use libc++?

jslee02, 2017-05-23:
I followed what you did but couldn't reproduce the errors. Could you make sure if the earlier versions of DART don't conflict? Also, are you using the [`CMakeLists.txt`](https://github.com/dartsim/dart/blob/master/tutorials/CMakeLists.txt) in the tutorials directory?

jyf588, 2017-05-25:
I think I solved the issue: indeed I have to use libc++ instead of libstdc++, as I mentioned above.

Adding SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libc++ -std=c++11") solved the problem.

Do you know why this only happens on my machine? Thanks.

PR #889 (2017-05-21): Use CONFIG mode to find DART in examples and tutorials
`find_package(DART)` uses [`FindDart.cmake`](https://github.com/Kitware/CMake/blob/master/Modules/FindDart.cmake) of the [DART programming language](https://www.dartlang.org/) by default. This PR changes the DART examples and tutorials to use `CONFIG` mode instead to find our DART using [`DARTConfig.cmake`](https://github.com/dartsim/dart/blob/master/cmake/DARTConfig.cmake.in).

This PR resolves #888.